espressif/esp-iot-bridge #6: "gateway/priv_src" directory disappeared?

zhuangqingshu, 2022-05-28:
Work on master branch
"gateway/priv_src" directory not found
But this directory is mentioned in "CMakeLists.txt" file
I need to use the "gateway_eth.c" file in this directory

components\gateway\CMakeLists.txt
```
set(srcs "src/gateway_common.c") set(requires "") set(include_dirs "include") set(priv_includes "priv_inc") set(priv_srcs "")
if (CONFIG_GATEWAY_EXTERNAL_NETIF_STATION OR CONFIG_GATEWAY_DATA_FORWARDING_NETIF_SOFTAP) list(APPEND srcs "src/gateway_wifi.c") endif()
if (CONFIG_GATEWAY_EXTERNAL_NETIF_MODEM) list(APPEND srcs "src/gateway_modem.c")
    if(CONFIG_GATEWAY_MODEM_USB) list(APPEND srcs "src/modem_board_usb.c") endif()
    if(CONFIG_GATEWAY_MODEM_CUSTOM_BOARD) list(APPEND srcs "src/modem_board_sim7600.c") endif() endif()
if (CONFIG_GATEWAY_DATA_FORWARDING_NETIF_USB) list(APPEND priv_srcs "priv_src/gateway_usb.c") endif()
if (CONFIG_GATEWAY_EXTERNAL_NETIF_ETHERNET OR CONFIG_GATEWAY_DATA_FORWARDING_NETIF_ETHERNET) list(APPEND priv_srcs "priv_src/gateway_eth.c") endif()
if (CONFIG_GATEWAY_DATA_FORWARDING_NETIF_SDIO) list(APPEND priv_srcs "priv_src/gateway_sdio.c") endif()
if (CONFIG_GATEWAY_DATA_FORWARDING_NETIF_SPI) list(APPEND priv_srcs "priv_src/gateway_spi.c") endif()
```

x893, 2022-05-28:
Also no file gateway_spi.c

if (CONFIG_GATEWAY_DATA_FORWARDING_NETIF_SPI)
    list(APPEND priv_srcs "priv_src/gateway_spi.c")
endif()

xcguang, 2022-05-30:
[image]

x893, 2022-06-02:
Thanks you very much.
Now all works fine.

closed 2022-10-27 by xcguang