espressif/esp-iot-bridge #7: How to use ethap0 ?

x893, 2022-06-02:
Hello,
i use esp-gateway with ESP32-S2 and RasPi over SPI.
Now i see 2 interfaces in RasPi: ethsta0 (192.168.5.2) and ethap0 (without ip).
ESP32-S2 successfully connect to WiFi-router and RasPi can access to internet over ESP-Gateway.
Also i can connect to ESP-Gateway access point (192.168.4.2) and see web page (192.168.4.1) with settings for SSID and password.
How can i route to RasPi over ESP-Gateway network ?
I think that ethap0 can connect to ESP-Gateway network (192.168.4.x) but after set ip address to ethap0 (192.168.4.100) - no ping and no routing from connected computer.

```
            (192.168.5.2)
           - ethsta0    +               (192.168.1.26)
RasPi      +            +-- SPI -- ESP32-S2 --> External WiFi
(bulleyes) - ethap0 (?) +            |                 ^ (192.168.1.20)
                                     +-- ESP-Gateway <-+- Computer
                                       (192.168.4.1)  (192.168.4.2)
```
So i need access from Computer to RasPi.
Can you help me with configuration ?

Thanks in advance
Mitek

tswen, 2022-06-02:
ethap0 is actually useless and can be ignored.
If you want to make RasPi(192.168.5.2) communicate with Computer(192.168.4.2) through ESP-Gateway, you need to apply the following patch under _**esp-idf/components/lwip/lwip**_:
```c
diff --git a/src/core/ipv4/ip4.c b/src/core/ipv4/ip4.c
index c616f73f..46f8bab8 100644
--- a/src/core/ipv4/ip4.c
+++ b/src/core/ipv4/ip4.c
@@ -383,8 +383,10 @@ ip4_forward(struct pbuf *p, struct ip_hdr *iphdr, struct netif *inp)

 #if ESP_LWIP
 #if IP_NAPT
-  if (ip_napt_forward(p, iphdr, inp, netif) != ERR_OK)
-    return;
+  if (!ip4_netif_exist(ip4_current_dest_addr(), ip4_current_dest_addr())) {
+    if (ip_napt_forward(p, iphdr, inp, netif) != ERR_OK)
+      return;
+  }
 #endif
 #endif /* ESP_LWIP */

```
The patch has not been merged into idf/master, so it needs to be applied manually.

Test on: 
- ESP-Gateway/master:
c51cb2a

- ESP-IDF/master:
9d9197c7c6

x893, 2022-06-02:
Thanks,
I'll try

So for Computer to RasPi i need change ip mask on Computer to
255.255.254.0
so 192.168.5.2 route to 192.168.4.1 interface ?

x893, 2022-06-02:
Hello,
i apply changes to **ip4.c**
ESP32-S2
I (1748) ip select: IP Address:192.168.5.1
I (1752) ip select: GW Address:192.168.5.1
I (1756) ip select: NM Address:255.255.255.0 (or need 255.25.254.0 also ?)
and connected to external WiFi router (192.168.1.26)

Computer (connect to ESP-Gateway) 192.168.4.10/255.255.254.0

RasPi (connect via SPI) 192.168.5.2/255.255.254.0

**RaspPi**
ping google.com - success
ping 192.168.5.1 (ESP32) - failed
ping 192.168.4.10 (Computer) - failed

**Computer**
ping 192.168.4.1 (ESP32) success
ping 192.168.5.2 (RasPi) - failed

May be some incorrect ?

tswen, 2022-06-06:
NM address need be 255.255.255.0
```c
I (698) wifi:mode : null
I (699) ip select: IP Address:192.168.4.1
I (699) ip select: GW Address:192.168.4.1
I (700) ip select: NM Address:255.255.255.0
I (704) gateway_wifi: IP Address:192.168.4.1
Add netif ap with c51cb2a(commit id)
I (712) gateway_common: netif list add success
I (718) wifi:mode : softAP (84:f7:03:80:00:fd)
I (722) wifi:Total power save buffer number: 16
I (726) wifi:Init max length of beacon: 752/752
I (730) wifi:Init max length of beacon: 752/752
I (1316) wifi:Total power save buffer number: 16
I (1319) gateway_wifi: softap ssid: ESP_Gateway password: 12345678
softap_netif 0x3fceb550 esp_gateway_create_softap_netif 263******************
Add netif sta with c51cb2a(commit id)
I (1327) gateway_common: netif list add success
I (1333) wifi:mode : sta (84:f7:03:80:00:fc) + softAP (84:f7:03:80:00:fd)
I (1339) wifi:enable tsf
I (1342) gateway_wifi: Found ssid **********
I (1347) gateway_wifi: Found password **********
station_netif 0x3fcec71c esp_gateway_create_station_netif 188******************
I (1359) ip select: IP Address:192.168.5.1
I (1364) ip select: GW Address:192.168.5.1
I (1368) ip select: NM Address:255.255.255.0
I (1373) mac select: MAC 84:f7:03:81:01:fd
```

tswen, 2022-06-27:
Hi, May I know if there is any update? If the issue still exists, please let us know.

closed 2022-10-27 by xcguang