espressif/esp-iot-bridge #8: 老板，可以增加esp-ide的.preject 文件吗？想用esp-ide 调试，vscode 调试信息不全

tigeratgithub, 2022-06-27:
老板，可以增加esp-ide的.preject 文件吗？想用esp-ide 编译和调试
当前版本好像不能用ESP-IDE编译和调试，多谢了

tswen, 2022-06-27:
如果有正常的 IDF 环境（release/v4.4 release/v5.0，建议使用 release/v4.4）那么都是可以正常编译的
```c
idf.py set-target esp32
idf.py build
idf.py -p <port> flash monitor
```
你可以先编译 IDF 下面的 demo 验证你的环境是否正常，如果还不行，可否详细描述目前遇到的问题。

tigeratgithub, 2022-06-27:
我下载的Espressif-IDE 2.5.0 with ESP-IDF v4.4，示例工程在ESP-IDF下可以编译下载的。esp-gateway工程在ESP-IDF 环境下也可以编译，下载。但在espressif-IDE中不能编译，提示确实 ymal文件。这个也可能是espressif-IDE 2.5的问题，你们可以打开试试

tigeratgithub, 2022-06-28:
支持C++的操作：
1. 在app_main.c中增加如下代码
``` C
    extern "C" {
        void app_main();
    }
```
2. rename app_main.c to main.cpp
3. idf.py menuconfig
4. idf.py build
5. 编译出错：
```
../main/main.cpp:90:5: warning: missing initializer for member 'esp_timer_create_args_t::dispatch_method' [-Wmissing-field-initializers]
../main/main.cpp:90:5: warning: missing initializer for member 'esp_timer_create_args_t::skip_unhandled_events' [-Wmissing-field-initializers]
[1097/1099] Linking CXX executable esp-gateway.elf
FAILED: esp-gateway.elf
cmd.exe /C "cd . && D:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\esp-gateway.elf.rsp -o esp-gateway.elf  && cd ."
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.cpp.obj):(.literal.app_main+0x10): undefined reference to `esp_gateway_create_all_netif()'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.cpp.obj): in function `app_main':
d:\esp32_project\esp-gateway\build/../main/main.cpp:117: undefined reference to `esp_gateway_create_all_netif()'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

closed 2022-06-28 by tigeratgithub