espressif/esp-iot-bridge #9: Not connection over USB to SIM7600E

RFgermany, 2022-08-22:
Hello,
first of all, very cool project Here! I super excited to get it working :)
There is the Issue that I can not get the CDC connection to work.
Sadly I can not find out what the problem is :-(
I hope you can help!
This is the Console log output:

`
I (2489) gateway_wifi: softap ssid: ESP_Gateway_70287d password: 12345678
W (2497) gateway_modem: Force reset 4g board
I (2502) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3011) gpio: GPIO[38]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (11011) gateway_modem: ====================================
I (11011) gateway_modem:      ESP 4G Cat.1 Wi-Fi Router
I (11012) gateway_modem: ====================================
D (11019) event: created event loop 0x3fceca00
I (11073) USB_HCDC: usb driver install succeed
I (11073) USB_HCDC: Waitting Device Connection
V (11073) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (11078) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x804
D (11087) intr_alloc: Connected src 38 to int 19 (cpu 0)
I (11122) USB_HCDC: USB Port=1 init succeed
I (11122) USB_HCDC: Waitting USB Connection
I (11122) USB_HCDC: Port power: ON
I (11374) USB_HCDC: line 268 HCD_PORT_EVENT_CONNECTION
I (11374) USB_HCDC: Resetting Port
I (11434) USB_HCDC: Port reset succeed
I (11434) USB_HCDC: Getting Port Speed
I (11434) USB_HCDC: Port speed = 1
I (11435) USB_HCDC: USB Speed: full-speed
D (11440) USB_HCDC: urb alloced
I (11443) USB_HCDC: Set Device Addr = 1
D (11449) USB_HCDC: line 616 Pipe: default HCD_PIPE_EVENT_URB_DONE
I (11454) USB_HCDC: Set Device Addr Done
D (11459) USB_HCDC: urb free
D (11462) USB_HCDC: urb alloced
I (11465) USB_HCDC: get device desc
D (11470) USB_HCDC: line 616 Pipe: default HCD_PIPE_EVENT_URB_DONE
I (11475) USB_HCDC: get device desc, actual_num_bytes:26
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x1e0e
idProduct 0x9001
bcdDevice 3.10
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
D (11505) USB_HCDC: urb free
D (11508) USB_HCDC: urb alloced
I (11511) USB_HCDC: get short config desc
D (11517) USB_HCDC: line 616 Pipe: default HCD_PIPE_EVENT_URB_DONE
I (11522) USB_HCDC: get config desc, actual_num_bytes:16
E (11528) USB_HCDC: Configuration descriptor larger than control transfer max length
D (11536) USB_HCDC: urb free
D (11539) USB_HCDC: urb alloced
I (11542) USB_HCDC: Set Device Configuration = 1
D (11548) USB_HCDC: line 616 Pipe: default HCD_PIPE_EVENT_URB_DONE
I (11554) USB_HCDC: Set Device Configuration Done
D (11559) USB_HCDC: urb free
I (11562) USB_HCDC: Creating bulk in pipe
I (11567) USB_HCDC: Creating bulk out pipe
V (11572) USB_HCDC: ST 0:
V (11575) USB_HCDC: ST 0:
V (11577) USB_HCDC: ST 0:
V (11580) USB_HCDC: ST 0:
W (11583) USB_HCDC: line 1030 Pipe: bulk_out HCD_PIPE_EVENT_ERROR_XFER
I (11588) USB_HCDC: Device Connected
D (11595) esp_netif_objects: esp_netif_add_to_list 0x3fcf2ff4
D (11600) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 2)
D (11609) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3fcaa238
D (11617) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (12124) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (12129) usb_modem_board: Current timeouts: 1 and errors: 0
D (12136) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:+++

E (17143) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
W (17143) esp_modem_dce: esp_modem_dce_generic_command(31): Command:+++ response timeout
D (17148) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (17656) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
D (17661) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (18169) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (18174) usb_modem_board: Current timeouts: 2 and errors: 0
I (18180) usb_modem_board: Restart to connect modem........
I (18187) usb_modem_board: modem_board_reset!
D (18192) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (18699) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (18704) usb_modem_board: Current timeouts: 3 and errors: 0
I (18710) usb_modem_board: modem_board_power_down!
D (18716) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (19223) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
D (19228) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (19736) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (19741) usb_modem_board: Current timeouts: 4 and errors: 0
I (19747) usb_modem_board: modem_board_power_down!
D (19753) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (20260) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
D (20265) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (20773) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (20778) usb_modem_board: Current timeouts: 5 and errors: 0
I (20784) usb_modem_board: modem_board_power_down!
D (20790) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (21297) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
D (21302) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (21810) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
E (21815) usb_modem_board: modem_board_start_up(136): sending sync failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4037c750
0x4037c750: _esp_error_check_failed at C:/Espressif/esp-idf/components/esp_system/esp_err.c:42

file: "../components/gateway/src/gateway_modem.c" line 238
func: esp_gateway_modem_init
expression: esp_modem_default_start(dte)

abort() was called at PC 0x4037c753 on core 0
`

tswen, 2022-08-22:
Hi
Are you only using **modem** and **softap**, can you try to run this [demo](https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module) and follow the [documentation](https://github.com/espressif/esp-iot-solution/blob/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module/README.md) above requirements to check whether the configuration is correct.

RFgermany, 2022-08-24:
Hi There!
I tired your suggestion and it throws exactly the same behavior. I checked that the Module works, by connecting it over USB to my Windows PC and there I can interface it. It appears to be working in general then.

RFgermany, 2022-08-24:
I went through all listest endpoint combinations and had a slightly different report on 0x84/0x03 and 0x86/0x04:
But still no working connetion...

I (13470) USB_HCDC: Device Connected
E (13971) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (13976) modem_board: Current timeouts: 1 and errors: 0
E (18982) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
W (18982) esp_modem_dce: esp_modem_dce_generic_command(31): Command:+++ response timeout
E (19487) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
E (19527) esp-modem-dte: esp_dte_handle_line(88): handle line failed
E (19529) esp-modem-dte: esp_dte_handle_line(88): handle line failed
W (19550) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT&W
...failed
I (19551) modem_board: Current timeouts: 0 and errors: 1
W (19555) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+CPIN?
...failed
W (19564) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+CPIN=1234
...failed
W (20575) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+CPIN?
...failed
E (20575) modem_board: modem_board_start_up(172): store_profile failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4037c5fc
0x4037c5fc: _esp_error_check_failed at C:/Espressif/esp-idf/components/esp_system/esp_err.c:42

tswen, 2022-08-24:
Hi
Please try again after modifying the usb port.

menuconfig PATH: Component config → ESP-MODEM → USB CDC endpoint address config

Modem USB CDC OUT endpoint address(0x05)
Modem USB CDC IN     endpoint address(0x88)

RFgermany, 2022-08-25:
Okay, I tried it again.
and prints Verbose:
......
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x8b   EP 11 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 8
                bInterval 32
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x8a   EP 10 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x6    EP 6 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
D (11868) USB_HCDC: urb free
D (11871) USB_HCDC: urb alloced
I (11874) USB_HCDC: Set Device Configuration = 1
D (11880) USB_HCDC: line 616 Pipe: default HCD_PIPE_EVENT_URB_DONE
I (11886) USB_HCDC: Set Device Configuration Done
D (11891) USB_HCDC: urb free
I (11894) USB_HCDC: Creating bulk in pipe
I (11899) USB_HCDC: Creating bulk out pipe
V (11904) USB_HCDC: ST 0:
V (11907) USB_HCDC: ST 0:
V (11909) USB_HCDC: ST 0:
V (11912) USB_HCDC: ST 0:
I (11917) USB_HCDC: Device Connected
D (11920) esp_netif_objects: esp_netif_add_to_list 0x3fcf2ff4
D (11925) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 2)
D (11934) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3fcaa238
D (11942) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:AT

E (12449) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (12454) usb_modem_board: Current timeouts: 1 and errors: 0
D (12461) esp_modem_dce: esp_modem_dce_generic_command(27): Sending command:+++
........

tswen, 2022-08-30:
Hi Florian,
Do you have any other 4G Cat.1 modules that we have already adapted? You can try the other ones. 
We will try to find a SIM7600E for testing recently.

RFgermany, 2022-08-30:
> Hi Florian, Do you have any other 4G Cat.1 modules that we have already adapted? You can try the other ones. We will try to find a SIM7600E for testing recently.

Sadly not, but I will continue to try. I have until next Tuesday to get it working. After that I will be on vacation for a month.

RFgermany, 2022-08-30:
I tried the -> ...\esp-idf\examples\peripherals\usb\host\cdc\cdc_acm_bg96 Example and there it works!
[image]

RFgermany, 2022-09-02:
I also got hold of a BG95-M3. Same behavior there. The cdc_acm_bg96 example works, but the esp-gateway can not get the AT communication working.