espressif/esp-iot-bridge #10: Can't connect to AP with SPI interface to RPi

x893, 2022-08-30:
Hi,
if i compile without SPI interface - i can connect to ESP Access Point and define External WiFi SSID and password. But if i add SPI interface to RasPi (without connection to RPi) - i can't connect to AP (no IP via DHCP). I see that something strange with IP address 192.168.5.1 (SoftAP and SPI slave) in second log.

I use ESP-IDF v4.4.2 and latest esp-gateway (and priv_src).
First log for esp-gateway **without** spi support and second **with** spi.

Thanks in advance

`
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.08.30 13:25:36 =~=~=~=~=~=~=~=~=~=~=~=
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x176c
load:0x4004c000,len:0xa80
load:0x40050000,len:0x2e5c
entry 0x4004c210
I (21) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (21) boot: compile time 13:02:02
I (21) boot: chip revision: 0
I (24) boot.esp32s2: SPI Speed      : 80MHz
I (29) boot.esp32s2: SPI Mode       : DIO
I (33) boot.esp32s2: SPI Flash Size : 4MB
I (38) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (62) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (69) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (77) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (84) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (92) boot:  5 ota_0            OTA app          00 10 00020000 001d0000
I (99) boot:  6 ota_1            OTA app          00 11 001f0000 001d0000
I (107) boot:  7 coredump         Unknown data     01 03 003c0000 00010000
I (114) boot:  8 log_info         Unknown data     01 fe 003d0000 00010000
I (122) boot:  9 reserved         Unknown data     01 ff 003e0000 00020000
I (129) boot: End of partition table
I (134) esp_image: segment 0: paddr=00020020 vaddr=3f000020 size=1fd44h (130372) map
I (168) esp_image: segment 1: paddr=0003fd6c vaddr=3ffc5620 size=002ach (   684) load
I (169) esp_image: segment 2: paddr=00040020 vaddr=40080020 size=7ccbch (511164) map
I (276) esp_image: segment 3: paddr=000bcce4 vaddr=3ffc58cc size=03190h ( 12688) load
I (279) esp_image: segment 4: paddr=000bfe7c vaddr=40022000 size=13620h ( 79392) load
I (301) esp_image: segment 5: paddr=000d34a4 vaddr=50000000 size=00010h (    16) load
I (311) boot: Loaded app from partition at offset 0x20000
I (311) boot: Disabling RNG early entropy source...
I (324) cache: Instruction cache : size 8KB, 4Ways, cache line size 32Byte
I (324) cpu_start: Pro cpu up.
I (344) cpu_start: Pro cpu start user code
I (344) cpu_start: cpu freq: 240000000
I (344) cpu_start: Application information:
I (347) cpu_start: Project name:     esp-gateway-4
I (353) cpu_start: App version:      v1.0.0-135-g4753c94
I (359) cpu_start: Compile time:     Aug 30 2022 13:16:56
I (365) cpu_start: ELF file SHA256:  954269d03015c3bd...
I (371) cpu_start: ESP-IDF:          v4.4.2
I (376) heap_init: Initializing. RAM available for dynamic allocation:
I (383) heap_init: At 3FFCC928 len 0002F6D8 (189 KiB): D/IRAM
I (389) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (395) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (402) spi_flash: detected chip: generic
I (406) spi_flash: flash io: dio
I (414) cpu_start: Starting scheduler on PRO CPU.
I (428) wifi:wifi driver task: 3ffd36d0, prio:23, stack:6656, core=0
I (428) system_api: Base MAC address is not set
I (432) system_api: read default base MAC address from EFUSE
I (440) wifi:wifi firmware version: eeaa27d
I (442) wifi:wifi certification version: v7.0
I (446) wifi:config NVS flash: enabled
[... 174 lines cut ...]
I (1407) NETWORK_ADAPTER: *********************************************************************
I (1416) NETWORK_ADAPTER: Supported features are:
I (1421) NETWORK_ADAPTER: - WLAN over SPI
I (1426) SPI_DRIVER: Using SPI interface
I (1431) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1440) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2450) NETWORK_ADAPTER: Initial set up done
I (2450) gateway_spi: Network Adapter initialization success
Add netif SPI_Netif with 5f302bb(commit id)

I (2454) gateway_common: netif list add success
I (2459) gateway_spi: SPI IP Address:192.168.5.1
Add netif sta with 4753c94(commit id)

I (2468) gateway_common: netif list add success
I (2474) wifi:mode : sta (34:b4:72:6a:68:6e) + softAP (34:b4:72:6a:68:6f)
I (2480) wifi:enable tsf
I (2483) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
start web server

I (15387) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (15388) wifi:station: 68:94:23:f9:af:4d join, AID=1, bgn, 20
W (15407) wifi:<ba-add>idx:2 (ifx:1, 68:94:23:f9:af:4d), tid:0, ssn:0, winSize:64
I (15415) gateway_wifi: STA Connecting to the AP again...
`

tswen, 2022-08-31:
<html>
<body>

IDF Version |   | Note
-- | -- | --
ESP-IDF Release/v4.4 | Not Support | Only one network data forwarding interface can be selected
ESP-IDF Release/v5.0 | Support | Currently, SDIO and SPI interfaces cannot be selected at the same time

</body>
</html>

Please switch ESP-IDF to release/v5.0 branch and try again.

x893, 2022-08-31:
Hello, i switch to v5.0 and AP works. Now port host driver for Zynq 7000 linux. Need time to check via SPI.

x893, 2022-09-02:
Complete

closed 2022-09-02 by x893