espressif/esp-iot-bridge #14: Fail with `custom IP` AP insist to assign 192.168.4.1

ldab, 2022-09-07:
I made the following change to show the problem on `gateway_common.c` and using IDF v4.4.2

[image]

The AP insist to stay in the `192.168.4.1` range as indicated on the log below:

```
I (770) ip select: IP Address:192.168.7.1
I (770) ip select: GW Address:192.168.7.1
I (770) ip select: NM Address:255.255.255.0
I (775) gateway_wifi: IP Address:192.168.4.1
```

tswen, 2022-09-08:
Hi
Please apply the patch below and try again.
```c
diff --git a/components/gateway/src/gateway_wifi.c b/components/gateway/src/gateway_wifi.c
index 8d8cb6c..176ae13 100644
--- a/components/gateway/src/gateway_wifi.c
+++ b/components/gateway/src/gateway_wifi.c
@@ -110,7 +110,7 @@ static void wifi_event_ap_start_handler(void *arg, esp_event_base_t event_base,
 
     if (netif) {
         if (esp_gateway_softap_dhcps) {
-            ESP_ERROR_CHECK(esp_netif_dhcps_stop(netif));
+            esp_netif_dhcps_stop(netif);
             esp_netif_dns_info_t dns;
             dns.ip.u_addr.ip4.addr = ESP_IP4TOADDR(114, 114, 114, 114);
             dns.ip.type = IPADDR_TYPE_V4;
@@ -244,15 +244,17 @@ esp_netif_t* esp_gateway_create_softap_netif(esp_netif_ip_info_t* ip_info, uint8
 
     wifi_netif = esp_netif_create_default_wifi_ap();
 
+    ESP_ERROR_CHECK(esp_netif_dhcps_stop(wifi_netif));
+
     if (ip_info) {
         ESP_ERROR_CHECK(esp_netif_set_ip_info(wifi_netif, ip_info));
     } else {
         if (enable_dhcps) {
             esp_gateway_netif_request_ip(&allocate_ip_info);
-            esp_netif_set_ip_info(wifi_netif, &allocate_ip_info);
+            ESP_ERROR_CHECK(esp_netif_set_ip_info(wifi_netif, &allocate_ip_info));
         }
     }
-    esp_netif_get_ip_info(wifi_netif, &netif_ip);
+    ESP_ERROR_CHECK(esp_netif_get_ip_info(wifi_netif, &netif_ip));
     ESP_LOGI(TAG, "IP Address:" IPSTR, IP2STR(&netif_ip.ip));
     esp_gateway_netif_list_add(wifi_netif, softap_netif_dhcp_status_change_cb);

```

ldab, 2022-09-08:
Confirmed to solve:

```
I (768) ip select: IP Address:192.168.0.27
I (769) ip select: GW Address:192.168.0.1
I (769) ip select: NM Address:255.255.255.0
I (773) gateway_wifi: IP Address:192.168.0.27
```

closed 2022-09-08 by ldab