espressif/esp-iot-bridge #17: Can't open Web page on 192.168.4.1

x893, 2022-10-19:
Hello
i use ESP32 with SDIO interface (not connected now to host).
Connect to ESP-Bridge AP from windows computer and receive 192.168.4.2 address.
Try open web page from 192.168.4.1 and see status
**192.168.4.1 refused to connect.**
Previous version esp-gateway open web page for wifi uplink settings.

How can i resolve it ?
Thanks
Mitek

tswen, 2022-10-19:
What are the versions of IDF and ESP-Bridge (branch and commit id)
In addition, how many data_forwarding_netifs are open in total, is there only softap, please make sure to open the BRIDGE_USE_WEB_SERVER configuration.
In addition, if there is another network card or wired connection on the computer side, it is recommended to disconnect the other network card and try again.

x893, 2022-10-19:
IDF 5.0 master
ESP-Bridge master
CONFIG_BRIDGE_USE_WEB_SERVER=y
SDIO not connected to ESP32-WROVER, only one computer connect to bridge
ping 192.168.4.1 works from computer

tswen, 2022-10-20:
Due to the particularity of the SDIO driver, SDIO needs to communicate with the MCU during initialization, otherwise it will be blocked. StartWebServer() is enabled after the SDIO netif is initialized and is not enabled, so the Web page cannot be accessed.
Do you have to configure the network first and then connect to SDIO?

x893, 2022-10-20:
OK, i connect SDIO later. Now in progress connection to Zynq 7000 system. Initially i use SPI interface and Web start normally. Notify about results later.
Thanks

xcguang, 2023-03-10:
Hello, Any update about this?

closed 2023-04-06 by xcguang