espressif/esp-iot-bridge #19: Failed to resolve component 'qrcode'

Leadrive, 2022-10-21:
idf.py menuconfig

report error
````
CMake Error at /home/lubuntu/esp/esp-idf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'qrcode'.
Call Stack (most recent call first):
  /home/lubuntu/esp/esp-idf/tools/cmake/build.cmake:227 (__build_resolve_and_add_req)
  /home/lubuntu/esp/esp-idf/tools/cmake/build.cmake:228 (__build_expand_requirements)
  /home/lubuntu/esp/esp-idf/tools/cmake/build.cmake:509 (__build_expand_requirements)
  /home/lubuntu/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:14 (project)
````

tswen, 2022-10-21:
Need to clone the submodule
`
git submodule update --init --recursive
`

Leadrive, 2022-10-21:
谢谢！

closed 2022-10-21 by Leadrive