espressif/esp-iot-bridge #20: 请教一下，如何进行Mesh-Lite测试？

Leadrive, 2022-10-21:
百度了下，没找到Mesh-Lite相关的资料。
我手上有好几块esp32的板子，要烧哪个固件，可以模拟wifi Mesh-Lite，与本项目进行通讯？

tswen, 2022-10-21:
esp-iot-bridge 根目录下面 menuconfig 打开 CONFIG_MESH_LITE_ENABLE 直接编译即可。
参考文档：https://github.com/espressif/esp-iot-bridge/blob/master/doc/Mesh-Lite.md

具体应用可以参考 [led_light](https://github.com/espressif/esp-iot-bridge/tree/master/examples/rainmaker/led_light)

Leadrive, 2022-10-21:
6. 在 Nove Home 中加入设备
一、打开 Nove Home，APP 自动搜索到待配网设备
二、添加对应的设备
三、输入配网信息
**第三步，为何是选的直接连路由器的WIFI？选esp网关SoftAP，连不上。**

四、点击配对和连接进行配网
**这里为何还要使用BLE配网？**

挺难配的，要尝试好几次才配上。效果比BLE MESH差太多。

另外，请教一下，不用NOVA HOME，自己通过调用API方式实现配网，能否出个简单的说明？

谢谢！

Leadrive, 2022-10-21:
MESH-Lite跟MDF，二者有什么区别呢？

问题多多，主要是刚接触这个，打扰了，不好意思。

Leadrive, 2022-10-22:
我把ESP-Bridge关掉，设备配网时，直接连的路由器ABC，也能配上网。
那这个ESP-Bridge支持Mesh-Lite，体现在哪方面呢？
好像不要ESP-Bridge，也能组成Mesh网。

Leadrive, 2022-10-22:
**单个 Bridge 设备最多支持 10 个 Station 同时连接，多个 Station 设备共享带宽。**

README中这句话，Station是指的是WIFI路由器吗？意思是一个Bridge可以同时连10个路由器上网？

tswen, 2022-10-23:
1. 配网可以通过 [webserve 或者 Wi-Fi provisioning](https://github.com/espressif/esp-iot-bridge#7-network-configuration)
2. Nova home 配网时填写的 ssid 均是路由器 ssid，后接入的设备会自动加入组网，只不过会存储路由器信息防止找不到合适的父节点
3. Mesh-Lite 内部会有组网管理的机制，Msh-Lite 和 ESP-WIFI-MESH 网络拓扑类似，都是 softap+station 形式，只不过实现形式略有不同，Mesh-Lite 和 ESP-WIFI-MESH 不可同时使用。

closed 2022-10-23 by Leadrive