espressif/esp-iot-bridge #21: 能否结合MDF中的MQTT例子？

Leadrive, 2022-10-23:
~/esp/esp-mdf/examples/function_demo/mwifi/mqtt_example
这个例子，正是我业务所需要的，能否结合这个例子，给个example？

谢谢！

tswen, 2022-10-23:
https://github.com/espressif/esp-iot-bridge/tree/master/examples/rainmaker/led_light

Mesh-Lite 相对于 ESP-WiFi-Mesh 最大的优点就是对于每个设备节点都有属于自己的 IP，可以独立上网，包括访问 MQTT，跟普通设备连接路由器上网应用开发是几乎没有区别的。

上面的例子连接 ESP-Rainmaker 云可以作为参考

Leadrive, 2022-10-23:
@tswen 
非常感谢你的回复，让我对Mesh-Lite多了些了解。
我很想用上这个，但参考led_light，我根本改不出一个纯粹的，连接自己MQTT服务器的例子。
程序中用到了众多esp_rmaker_函数，无从下手。

我想要一个象MDF中的MQTT例子：
1、配网写死的
类似这样
#
# Example Configuration
#
CONFIG_ROUTER_SSID="xxx"
CONFIG_ROUTER_PASSWORD="12345678"
CONFIG_MESH_ID="123456"
CONFIG_MESH_PASSWORD="MESH_PASSWORD"
CONFIG_MQTT_URL="mqtt://mqtt.eclipse.org"

2、收、发MQTT消息

真的很需要，看能否抽时间弄个DEMO，谢谢！！

tswen, 2022-10-23:
那你直接在根目录编译就 ok，只需要 menuconfig 开启 Mesh-Lite，每个设备都有独立的 IP，连接 MQTT 服务器直接仿照 IDF 里面的 example 写就行了

Leadrive, 2022-10-23:
@tswen 按你说的，已可以收/发信息，非常开心，终于有小小收获，非常感谢。

还有几个疑问请教一下：
1 如何设置MESH_ID, MESH_PASSWORD，只有符合的，才入网。

2 根节点必须要有连接外网的能力吗？如果有此要求，如何确保只有连接外网能力的，才能成为根节点？
如果没此要求，我的理解是，随便哪个设备，都可以连接外网吗？如果是这样，所有设备都配上路由器SSID，
都能上网，岂不又变成加重路由器负担？

3 如果一个组网内，有一个设备以太网，有一个设备是4G，有一个设备是WIFI，组内是如何确定哪种方式上网？

4 如果一个设备，既有以太网，又配了WIFI，如何采用优先 以太网>WIFI？

tswen, 2022-10-24:
1. mesh_id 可以在 menuconfig 里面配置，不知道你指的 mesh_password 是不是 softap 的 password，都可以在 menuconfig->ESP Wi-Fi Mesh Lite 里面配置
2. 默认就是只有连接了路由器之后才会成为 level 1 根节点，因为 CONFIG_JOIN_MESH_IGNORE_ROUTER_STATUS 配置项是默认不开启的
3. 设备上网优先级可以参考这个头文件配置
https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_netif/include/esp_netif_defaults.h#L29

Leadrive, 2022-10-24:
我看MDF的例子，是检查ID+PASSWORD的，只检查ID，也没什么问题。

closed 2022-10-24 by Leadrive