espressif/esp-iot-bridge #24: ESP_Mesh_Lite_Comm: Error occurred during sending: errno Not enough space

Leadrive, 2022-10-27:
````
I (25496) ESP_Mesh_Lite_Comm: Message sent: {"monkey":7217,"type":"volunteer_req","seq":203035214,"payload":{"router":-75,"mac":"24:0a:c4:58:00:8d","votes":[1,0,0,0,0]}}
W (26394) wifi:<ba-add>idx:2 (ifx:1, 94:e6:86:37:9f:3c), tid:0, ssn:0, winSize:64
I (26451) ESP_Mesh_Lite_Comm: rx data: {"monkey":3413043165,"type":"argot_req","seq":1452031047,"payload":{"argot":2147483647,"mac":"���7�<�?���(��?\u0004"}}
I (26458) ESP_Mesh_Lite_Comm: response data: {"monkey":23408125,"type":"argot_rsp","seq":1452031047,"payload":{"argot":2147483647}}
I (26469) ESP_Mesh_Lite_Comm: free heap: 159924
I (26496) ESP_Mesh_Lite_Comm: Message sent: {"monkey":7217,"type":"volunteer_req","seq":203035214,"payload":{"router":-75,"mac":"24:0a:c4:58:00:8d","votes":[1,0,0,0,0]}}
I (27496) ESP_Mesh_Lite_Comm: Message sent: {"monkey":7217,"type":"volunteer_req","seq":203035214,"payload":{"router":-75,"mac":"24:0a:c4:58:00:8d","votes":[1,0,0,0,0]}}
I (28496) ESP_Mesh_Lite_Comm: Message sent: {"monkey":7217,"type":"volunteer_req","seq":203035214,"payload":{"router":-75,"mac":"24:0a:c4:58:00:8d","votes":[1,0,0,0,0]}}
E (28501) ESP_Mesh_Lite_Comm: Error occurred during sending: errno Not enough space
I (29496) ESP_Mesh_Lite_Comm: Message sent: {"monkey":7217,"type":"volunteer_req","seq":203035214,"payload":{"router":-75,"mac":"24:0a:c4:58:00:8d","votes":[1,0,0,0,0]}}
E (29501) ESP_Mesh_Lite_Comm: Error occurred during sending: errno Not enough space
````
只有重启才行

tswen, 2022-10-28:
请详细描述下遇到的问题，什么场景

Leadrive, 2022-11-03:
暂时先关闭了

closed 2022-11-03 by Leadrive