espressif/esp-iot-bridge #26: 4G Router using SIM7600 and ESP Wroom32

RenjithBeginow, 2022-11-22:
Hi myself Renjith.
I am trying to set a 4G router using SIM7600 and ESPwroom32.
I have configured the details using menuconfig, but Iam getting an error and ESP is getting force restart.

I (62639) bridge_modem: Modem PPP Started
I (62734) bridge_modem: IP event! 6
I (62735) esp-netif_lwip-ppp: Connected
I (62735) bridge_modem: Modem Connected to PPP Server
I (62738) bridge_modem: ppp ip: 25.43.7.145, mask: 255.255.255.255, gw: 10.64.64.64
I (62747) bridge_modem: Main DNS: 49.45.0.1
I (62752) bridge_modem: Backup DNS: 0.0.0.0
**E (62758) bridge_modem: PPP netif event = 0
E (62761) bridge_modem: Just Force restart!**

Please help me to sort it out.

tswen, 2022-11-22:
Hi @RenjithBeginow 
What is the version of idf you are using, you can delete this esp_restart() first, this will not be affected.

RenjithBeginow, 2022-11-22:
Hi tswen,
IDF version ESP-IDF v5.1-dev-1626-g4b6d9c8ad3-dirty,
OK let me try by deleting the esp_restart()

tswen, 2022-11-22:
If you will only use a data forwarding netif, such as softap, it is recommended that you use idf release/4.4

RenjithBeginow, 2022-11-22:
Hi,
After deleting the esp_restart() function, I am able to connect device to the softAP of ESP and the device is getting internet now.
But I have noticed that the speed is too too slow, is there anything i have to do in the code, also when i debug with the serial port of the ESP is giving result.
**W (857566) esp-modem-dte: Rx Break**
Is it due to the buffer size or something like that?

RenjithBeginow, 2022-11-23:
Thanks for the support tswen.
Let me try with idf release4.4 and I will let you know the feed back.

RenjithBeginow, 2022-11-24:
Hi, 
I have tried with release/v4.4.3, Iam getting internet but the speed is only almost 15-20Kbps.
Also Iam getting this return frequently while trying to connect with internet.
**esp-modem-dte: Rx Break**

RenjithBeginow, 2022-12-01:
Hi Tswen,
Can I have your mail ID?

tswen, 2022-12-01:
This communication rate depends on your uart baud rate, if you need high throughput, it is recommended to use the usb interface

RenjithBeginow, 2022-12-01:
Hi Tswen,
Can you help me to implement with USB, I have ESP32S3 and SIMCOM A7600 modem.
While communicating with USB its showing waiting for USB port, while connecting A7600 to PC it will show 4USB ports, so will I have to do anything in the ESP32 code to tell ESP that this is the port.

tswen, 2022-12-01:
If you only need to implement 4G -> AP, it is recommended that you use this [demo](https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module) , Is it because the 4G module has insufficient power supply that the usb connection is not detected?

RenjithBeginow, 2022-12-02:
Hi,
Iam getting debug result as given below, and its not going forward. Waiting

I (0) cpu_start: Starting scheduler on APP CPU.
I (3403) 4g_main: ====================================
I (3403) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (3403) 4g_main: ====================================
I (3409) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3418) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (3428) modem_board: Force reset modem
I (3432) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (3441) modem_board: Resetting modem using io=10, level=0
W (3647) modem_board: Waiting for modem initialize ready
I (8677) ESP_USBH: USB Port=1 init succeed, fifo strategy=0
I (8677) USB_HCDC: CDC task start
I (8677) ESP_USBH: usb port start succeed
I (8681) USB_HCDC: usb driver install succeed
I (8686) USB_HCDC: Waiting CDC Device Connection
I (8691) ESP_USBH: USB Processing Start
I (8695) ESP_USBH: Waiting USB Connection
I (8700) ESP_USBH: Port power: ON Succeed

Can u please check this.

tswen, 2022-12-05:
The program running here means that the usb host has been initialized and is waiting for the connection of the usb device. If it is not confirmed that the usb has been inserted, please confirm whether the modem module is reset in time, or whether the power supply to the 4G module is abnormal.

https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module#user-guide

RenjithBeginow, 2022-12-07:
Hi, 
I have checked the power supply and connections, but still iam getting the same output.
**I (24) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (25) boot: compile time 12:40:52
I (25) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (92) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=19dcch (105932) map
I (121) esp_image: segment 1: paddr=00029df4 vaddr=3fc98d30 size=041c0h ( 16832) load
I (124) esp_image: segment 2: paddr=0002dfbc vaddr=40374000 size=0205ch (  8284) load
I (129) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=7a95ch (502108) map
I (211) esp_image: segment 4: paddr=000aa984 vaddr=4037605c size=12ccch ( 77004) load
I (226) esp_image: segment 5: paddr=000bd658 vaddr=50000000 size=00010h (    16) load
I (235) boot: Loaded app from partition at offset 0x10000
I (235) boot: Disabling RNG early entropy source...
I (247) cpu_start: Pro cpu up.
I (247) cpu_start: Starting app cpu, entry point is 0x403755d8
I (0) cpu_start: App cpu up.
I (261) cpu_start: Pro cpu start user code
I (261) cpu_start: cpu freq: 240000000
I (261) cpu_start: Application information:
I (263) cpu_start: Project name:     cdc_4g
I (268) cpu_start: App version:      1cd3f3d
I (273) cpu_start: Compile time:     Dec  7 2022 10:55:19
I (279) cpu_start: ELF file SHA256:  a365eb58f5500372...
I (285) cpu_start: ESP-IDF:          v4.4.3-dirty
I (291) heap_init: Initializing. RAM available for dynamic allocation:
I (298) heap_init: At 3FCA05E8 len 00049128 (292 KiB): D/IRAM
I (304) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (311) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (317) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (324) spi_flash: detected chip: generic
I (328) spi_flash: flash io: qio
W (332) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (346) sleep: Configure to isolate all GPIO pins in sleep state
I (352) sleep: Enable automatic switching of GPIO sleep configuration
I (359) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (3404) 4g_main: ====================================
I (3404) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (3404) 4g_main: ====================================
I (48410) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (48410) gpio: GPIO[1]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (48418) gpio: GPIO[40]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (48427) modem_board: Force reset modem
I (48432) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (48441) modem_board: Resetting modem using io=26, level=0
W (48647) modem_board: Waiting for modem initialize ready
I (53677) ESP_USBH: USB Port=1 init succeed, fifo strategy=0
I (53677) ESP_USBH: usb port start succeed
I (53677) USB_HCDC: CDC task start
I (53677) USB_HCDC: usb driver install succeed
I (53686) USB_HCDC: Waiting CDC Device Connection
I (53677) ESP_USBH: USB Processing Start
I (53696) ESP_USBH: Waiting USB Connection
I (53701) ESP_USBH: Port power: ON Succeed
** 

Also when I connect the gsm module reset to ny IO the modem is not at all getting ON, So I have only given the powerkey pin. After modem is getting ON I will start the ESP32, is there any problem by doing this.

RenjithBeginow, 2022-12-08:
Hi,
Is there anything else i need to do or taken care in order to enable USB. I have checked power supply and the USB cable iam using, but still its not moving further. Still getting the same output.

leeebo, 2022-12-08:
Hi @RenjithBeginow, the log shows no device appears, please double check your hardware connection, if you just insert a USB disk, can it goes further?

RenjithBeginow, 2022-12-08:
Thanks leebo, 
I will check and let you know.

RenjithBeginow, 2022-12-08:
Hi leeebo,
As u said I have tried with a USB drive connected to the usb port of esp32s3, but still its not proceeding to further.
Its waiting "ESP_USBH: Port power: ON Succeed" here itself.

RenjithBeginow, 2022-12-12:
Hi,
Can u help to sort it out. I have got stuck here.

RenjithBeginow, 2022-12-13:
Hi @leeebo leeebo & @tswen ,
Please help for the development.

leeebo, 2022-12-13:
@RenjithBeginow Hi, if the USB drive can not be detected also, the most likely issue is the power supply. Which development board do you use? Does the USB interface have reliable 5V output?

RenjithBeginow, 2022-12-14:
Hi,
I have made the power of ESP also from a 5v power supply to esp32s3, still the program is waiting there itself, i am not able to debug the problem.

m5it, 2022-12-15:
you achieve to connect to device and upload you sketch or binary or your program you are trying to run? maybe you need to hold down boot button until it get connected?