espressif/esp-iot-bridge #34: Packet jam occurs on Quectel EC25 and ESP32-S3 with USB connection.

jitenshap, 2023-01-27:
I trying Quectel EC25 and ESP32-S3 with USB connection.
I have a successful PPP connection within the 4g_hotspot example with the config shown below.

CONFIG_BRIDGE_MODEM_USB_VID=0x2C7C
CONFIG_BRIDGE_MODEM_USB_PID=0x125
CONFIG_BRIDGE_MODEM_USB_INTERFACE_NUMBER=2

However, I connected with ESP32-S3 via Wi-Fi with my PC or smartphone, but I was unable to view the content in the web browser.
curl "http://httpbin.org/get" command executed successfully and received HTTP response.
But, Nothing can be displayed in the web browser.
I did a packet capture of the Wi-Fi with WireShark. There I found a lot of TCP packet retransmissions.
I've attached the log with LWIP's ppp logging enabled.
From the original example, I changed the DNS server to 8.8.8.8 and the IP address to 192.168.200.1.
I also have PSRAM is enabled.

[ec25_bridge_log.txt](https://github.com/espressif/esp-iot-bridge/files/10519694/ec25_bridge_log.txt)

tswen, 2023-02-01:
Could it be that the signal of the 4G module is not very good, or is it caused by the interference of the wireless environment, can the device access other web pages except http://httpbin.org/get, such as Google?

jitenshap, 2023-02-12:
Network health is not an issue. CSQ shows a value of over 20 and my ESP32-S3 and PC are on the same desk.
I found to there is no problem with HTTP communication, but when using HTTPS communication.
"pppos_netif_output[2]: output failed proto=0x21, len = 569"
error is displayed.

jitenshap, 2023-02-12:
My EC25 is able to communicate over HTTPS with OpenWRT's PPP connection without any problems.
Are you able to communicate via HTTPS with modems officially supported by esp-iot-bridge?

tswen, 2023-03-10:
We have tried a variety of modems, all of which can access HTTPS normally, you can also use this [demo](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module) to test.

jitenshap, 2023-03-28:
Tried again with another SIM card I have.
It was my misunderstanding that HTTPS communication was not possible.
In curl I saw the HTTP response to the curl "https://www.google.com" command.
However, web browsers cannot display any pages.
A website's HTML typically loads multiple resources.
The esp-iot-bridge seems to jam packets when many HTTP requests are processed at the same time.