espressif/esp-iot-bridge #37: [Tracking issue] USB to ethernet bridge for mesh-lite network

redfast00, 2023-03-04:
Currently, there is code for bridging esp-mesh-lite over wifi, so that other (non-espressif) devices can connect to one of the AP's provided by the meshed ESP32 devices, and they'll have internet as well. There's also an option to do this for USB as well (using the TinyUSB stack), so that the device the ESP32 is plugged into can access the network as well.

Unfortunately, the latter is behind the CONFIG_GATEWAY_DATA_FORWARDING_NETIF_USB option in the menuconfig, which is marked as `Use USB interface to provide network data forwarding for other devices (Please contact sales@espressif.com to apply for it)`. The source code for it is in a `priv_src` folder, which isn't public.

I've contacted sales to apply for this feature, but got a negative answer and was suggested to wait for the public version of this feature in the future. At the moment, no time estimate for when this will be completed was provided.

Would it be possible to implement this feature ourselves?

xcguang, 2023-03-07:
Hello, we will open these code soon

BraweProg, 2023-03-08:
Hello, I'm working on a ethernet-to-wifi-bridge and would highly appreciate to get priv_src/bridge_eth.c

Best Regards

redfast00, 2023-03-10:
👀 https://github.com/espressif/esp-iot-bridge/compare/b905253e18...028f7d2a93

Epic! Thank you so much :)

xcguang, 2023-03-10:
Hello all, The code is opend at https://github.com/espressif/esp-iot-bridge/commit/028f7d2a938377686a9bbe5962be21b2463d6187

closed 2023-03-10 by xcguang