espressif/esp-iot-bridge #38: Configuration descriptor larger than control transfer max length AIR780E and ESP32-S3 with USB connection

zsy-cn, 2023-03-06:
log
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1648
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2ef8
entry 0x403c9900
I (25) boot: ESP-IDF v5.0.1 2nd stage bootloader
I (25) boot: compile time 12:24:45
I (25) boot: chip revision: v0.1
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (72) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  3 factory          factory app      00 00 00010000 00100000
I (87) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (94) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (102) boot: End of partition table
I (106) boot: Defaulting to factory image
I (111) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=265ech (157164) map
I (148) esp_image: segment 1: paddr=00036614 vaddr=3fc96a00 size=03db0h ( 15792) load
I (151) esp_image: segment 2: paddr=0003a3cc vaddr=40374000 size=05c4ch ( 23628) load
I (159) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=90d1ch (593180) map
I (268) esp_image: segment 4: paddr=000d0d44 vaddr=40379c4c size=0ccc0h ( 52416) load
I (288) boot: Loaded app from partition at offset 0x10000
I (288) boot: Disabling RNG early entropy source...
I (299) cpu_start: Pro cpu up.
I (299) cpu_start: Starting app cpu, entry point is 0x40375348
0x40375348: call_start_cpu1 at /Users/a/esp/esp-idf-v5.0.1/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (313) cpu_start: Pro cpu start user code
I (314) cpu_start: cpu freq: 160000000 Hz
I (314) cpu_start: Application information:
I (317) cpu_start: Project name:     4g_hotspot
I (322) cpu_start: App version:      v1.0.0-184-gb49e745
I (328) cpu_start: Compile time:     Mar  6 2023 14:00:31
I (334) cpu_start: ELF file SHA256:  877f3a75940bac28...
I (340) cpu_start: ESP-IDF:          v5.0.1
I (345) cpu_start: Min chip rev:     v0.0
I (349) cpu_start: Max chip rev:     v0.99 
I (354) cpu_start: Chip rev:         v0.1
I (359) heap_init: Initializing. RAM available for dynamic allocation:
I (366) heap_init: At 3FC9E218 len 0004B4F8 (301 KiB): D/IRAM
I (372) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (379) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (385) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (393) spi_flash: detected chip: gd
I (396) spi_flash: flash io: dio
W (400) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (414) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (468) pp: pp rom version: e7ae62f
I (469) net80211: net80211 rom version: e7ae62f
I (471) wifi:wifi driver task: 3fced908, prio:23, stack:6656, core=0
I (473) system_api: Base MAC address is not set
I (478) system_api: read default base MAC address from EFUSE
I (491) wifi:wifi firmware version: 17afb16
I (491) wifi:wifi certification version: v7.0
I (493) wifi:config NVS flash: enabled
I (496) wifi:config nano formating: disabled
I (500) wifi:Init data frame dynamic rx buffer num: 32
I (505) wifi:Init management frame dynamic rx buffer num: 32
I (510) wifi:Init management short buffer num: 32
I (515) wifi:Init dynamic tx buffer num: 32
I (519) wifi:Init static tx FG buffer num: 2
I (523) wifi:Init static rx buffer size: 1600
I (527) wifi:Init static rx buffer num: 10
I (531) wifi:Init dynamic rx buffer num: 32
I (535) wifi_init: rx ba win: 6
I (538) wifi_init: tcpip mbox: 32
I (542) wifi_init: udp mbox: 6
I (546) wifi_init: tcp mbox: 6
I (550) wifi_init: tcp tx win: 5744
I (554) wifi_init: tcp rx win: 5744
I (558) wifi_init: tcp mss: 1460
I (562) wifi_init: WiFi IRAM OP enabled
I (566) wifi_init: WiFi RX IRAM OP enabled
I (572) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11
I (615) wifi:mode : null
I (616) ip select: IP Address:192.168.4.1
I (616) ip select: GW Address:192.168.4.1
I (617) ip select: NM Address:255.255.255.0
I (621) bridge_wifi: IP Address:192.168.4.1
Add netif ap with b49e745(commit id)
I (629) bridge_common: netif list add success
I (634) wifi:mode : softAP (f4:12:fa:de:a2:39)
I (641) wifi:Total power save buffer number: 16
I (642) wifi:Init max length of beacon: 752/752
I (647) wifi:Init max length of beacon: 752/752
W (652) bridge_modem: Force reset 4g board
I (656) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (6165) bridge_modem: Initializing esp_modem for the BG96 module...
I (6166) bridge_modem: Waiting for USB device connection...
E (6569) HUB: Configuration descriptor larger than control transfer max length
E (6570) HUB: Stage failed: CHECK_SHORT_CONFIG_DESC
```

zsy-cn, 2023-03-06:
CONFIG_BRIDGE_MODEM_USB_VID=0x19D1
CONFIG_BRIDGE_MODEM_USB_PID=0x1
CONFIG_BRIDGE_MODEM_USB_INTERFACE_NUMBER=2

tswen, 2023-03-10:
Please change the configuration item `CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE` to 1024

zsy-cn, 2023-03-13:
thank you, i have change the config item `CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE` to 1024
i got the other error panic
```
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759a8
0x403759a8: esp_restart_noos at /Users/a/esp/esp-idf-v5.0.1/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x164c
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2ef8
entry 0x403c9900
I (29) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (29) boot: compile time 13:37:51
I (29) boot: chip revision: v0.1
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (70) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (77) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  3 factory          factory app      00 00 00010000 00100000
I (92) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (99) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (107) boot: End of partition table
I (111) boot: Defaulting to factory image
I (116) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=26624h (157220) map
I (153) esp_image: segment 1: paddr=0003664c vaddr=3fc96a00 size=03db0h ( 15792) load
I (156) esp_image: segment 2: paddr=0003a404 vaddr=40374000 size=05c14h ( 23572) load
I (164) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=90da0h (593312) map
I (273) esp_image: segment 4: paddr=000d0dc8 vaddr=40379c14 size=0ccf8h ( 52472) load
I (293) boot: Loaded app from partition at offset 0x10000
I (293) boot: Disabling RNG early entropy source...
I (304) cpu_start: Pro cpu up.
I (304) cpu_start: Starting app cpu, entry point is 0x40375348
0x40375348: call_start_cpu1 at /Users/a/esp/esp-idf-v5.0.1/components/esp_system/port/cpu_start.c:142

I (283) cpu_start: App cpu up.
I (319) cpu_start: Pro cpu start user code
I (319) cpu_start: cpu freq: 160000000 Hz
I (319) cpu_start: Application information:
I (322) cpu_start: Project name:     4g_hotspot
I (327) cpu_start: App version:      1
I (331) cpu_start: Compile time:     Mar 13 2023 13:47:10
I (337) cpu_start: ELF file SHA256:  7fff187fd9125e22...
I (343) cpu_start: ESP-IDF:          v5.0.1-dirty
I (349) cpu_start: Min chip rev:     v0.0
I (353) cpu_start: Max chip rev:     v0.99 
I (358) cpu_start: Chip rev:         v0.1
I (363) heap_init: Initializing. RAM available for dynamic allocation:
I (370) heap_init: At 3FC9E218 len 0004B4F8 (301 KiB): D/IRAM
I (377) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (383) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (389) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (397) spi_flash: detected chip: gd
I (400) spi_flash: flash io: dio
W (404) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (418) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (474) bridge_common: esp-iot-bridge version: 0.3.0
I (475) pp: pp rom version: e7ae62f
I (476) net80211: net80211 rom version: e7ae62f
I (480) wifi:wifi driver task: 3fced908, prio:23, stack:6656, core=0
[... 54 lines cut ...]
0x42012cee: cdc_acm_transfers_allocate at /Users/a/Downloads/esp-iot-bridge-master/examples/4g_hotspot/managed_components/espressif__usb_host_cdc_acm/cdc_acm_host.c:577 (discriminator 5)

0x420134bf: cdc_acm_host_open_vendor_specific at /Users/a/Downloads/esp-iot-bridge-master/examples/4g_hotspot/managed_components/espressif__usb_host_cdc_acm/cdc_acm_host.c:808

0x4200ba33: CdcAcmDevice::open_vendor_specific(unsigned short, unsigned short, unsigned char, cdc_acm_host_device_config_t const*) at /Users/a/Downloads/esp-iot-bridge-master/examples/4g_hotspot/managed_components/espressif__usb_host_cdc_acm/include/usb/cdc_acm_host.h:298
 (inlined by) esp_modem::UsbTerminal::UsbTerminal(esp_modem_dte_config const*) at /Users/a/Downloads/esp-iot-bridge-master/examples/4g_hotspot/managed_components/espressif__esp_modem_usb_dte/esp_modem_usb.cpp:91

0x4200bad7: std::_MakeUniq<esp_modem::UsbTerminal>::__single_object std::make_unique<esp_modem::UsbTerminal, esp_modem_dte_config const*&>(esp_modem_dte_config const*&) at /Users/a/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/11.2.0/bits/unique_ptr.h:962
 (inlined by) esp_modem::create_usb_terminal(esp_modem_dte_config const*) at /Users/a/Downloads/esp-iot-bridge-master/examples/4g_hotspot/managed_components/espressif__esp_modem_usb_dte/esp_modem_usb.cpp:179

0x4200b144: esp_modem::create_usb_dte(esp_modem_dte_config const*) at /Users/a/Downloads/esp-iot-bridge-master/examples/4g_hotspot/managed_components/espressif__esp_modem_usb_dte/esp_modem_usb_api_target.cpp:17

0x4200ae46: esp_modem_new_dev_usb at /Users/a/Downloads/esp-iot-bridge-master/examples/4g_hotspot/managed_components/espressif__esp_modem_usb_dte/esp_modem_usb_c_api.cpp:19

0x42008b0f: esp_bridge_create_modem_netif at /Users/a/Downloads/esp-iot-bridge-master/components/iot_bridge/src/bridge_modem.c:175 (discriminator 13)

0x420081b7: esp_bridge_create_all_netif at /Users/a/Downloads/esp-iot-bridge-master/components/iot_bridge/src/bridge_common.c:346

0x42007fab: app_main at /Users/a/Downloads/esp-iot-bridge-master/examples/4g_hotspot/main/app_main.c:34 (discriminator 2)

0x420905bd: main_task at /Users/a/esp/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131

0x4037f9a1: vPortTaskWrapper at /Users/a/esp/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

```

zsy-cn, 2023-03-13:
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_LEDC_SUPPORT_APB_CLOCK=y
CONFIG_SOC_LEDC_SUPPORT_XTAL_CLOCK=y
CONFIG_SOC_LEDC_CHANNEL_NUM=8
CONFIG_SOC_LEDC_TIMER_BIT_WIDE_NUM=14
CONFIG_SOC_LEDC_SUPPORT_FADE_STOP=y
CONFIG_SOC_MPU_MIN_REGION_SIZE=0x20000000
CONFIG_SOC_MPU_REGIONS_MAX_NUM=8
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_LCDCAM_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_CACHE_SUPPORT_WRAP=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_RISCV_COPROC_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_USB_OTG_SUPPORTED=y
CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_PSRAM_DMA_CAPABLE=y
CONFIG_SOC_XT_WDT_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_HMAC_SUPPORTED=y
CONFIG_SOC_DIG_SIGN_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_MEMPROT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_APPCPU_HAS_CLOCK_GATING_BUG=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_ARBITER_SUPPORTED=y
CONFIG_SOC_ADC_FILTER_SUPPORTED=y
CONFIG_SOC_ADC_MONITOR_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=24
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=4
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=83333
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=611
[... 1751 lines cut ...]
# CONFIG_PPP_MPPE_SUPPORT is not set
# CONFIG_PPP_DEBUG_ON is not set
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

```

zsy-cn, 2023-06-13:
Already fixed

closed 2023-06-13 by zsy-cn

xiao-orien, 2023-08-09:
> thank you, i have change the config item `CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE` to 1024 i got the other error panic
>
> ```
> Rebooting...
> �ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0xc [...]

I also have the same problem, how to solve it。

zsy-cn, 2023-10-02:
update idf to V5.0.2

K372470, 2024-04-18:
I have an esp32s3 (its a bit burned down, but usb works, and it can be programmed throught programator), which works with MIDI keyboard ( so it works =P) , but when i try to get it work with another device, with the biggest packet around 512kb, i get the same error as you did.
When I try to set CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE, it changes nothing
Im coding using PlatformIO with arduino framework, but without any libraries (ex. usb_host_device_open())
Can you give me any tips to fix that, please?

K372470, 2024-04-27:
Does anyone know, is it possible to change CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE  property in arduino or its only for espidf?