espressif/esp-iot-bridge #39: USB bridge feature does not seem to compile:

redfast00, 2023-03-13:


tswen, 2023-03-14:
You can temporarily add the following patch to the code, and we will update the example for the NIC solution later.

```
diff --git a/components/iot_bridge/CMakeLists.txt b/components/iot_bridge/CMakeLists.txt
index 61e8593..6a0d290 100644
--- a/components/iot_bridge/CMakeLists.txt
+++ b/components/iot_bridge/CMakeLists.txt
@@ -30,7 +30,7 @@ if (CONFIG_BRIDGE_DATA_FORWARDING_NETIF_SDIO OR CONFIG_BRIDGE_DATA_FORWARDING_NE
 endif()
 
 if ("${IDF_TARGET}" STREQUAL "esp32s2" OR "${IDF_TARGET}" STREQUAL "esp32s3")
-    list(APPEND requires "esp_modem_usb_dte")
+    list(APPEND requires "esp_modem_usb_dte" "usb_device")
 endif()
 
 idf_component_register(SRCS "${srcs}"
diff --git a/examples/wifi_router/main/idf_component.yml b/examples/wifi_router/main/idf_component.yml
index 12e8b33..8923247 100644
--- a/examples/wifi_router/main/idf_component.yml
+++ b/examples/wifi_router/main/idf_component.yml
@@ -11,6 +11,9 @@ dependencies:
   web_server:
     path: components/web_server
     git: https://github.com/espressif/esp-iot-bridge.git
+  usb_device:
+    path: components/usb/usb_device
+    git: https://github.com/espressif/esp-iot-bridge.git
   # The "esp_modem_usb_dte" component is temporarily disabled in iot_bridge yml file,
   # as idf-component-manager doesn't support uploading components with "rules" entries to the registry.
   # So temporarily place the "esp_modem_usb_dte" "esp_modem" component under examples.

```

redfast00, 2023-06-01:
@tswen Would you accept a PR that includes your patch?

tswen, 2023-06-08:
Hi @redfast00 , there is an internal MR under review and will be released soon.

chedim, 2024-02-10:
how soon, tho?

tswen, 2024-02-17:
The current example can already use USB netif normally and does not require any modification to the components.