espressif/esp-iot-bridge #40: DHCP server on USB interface

redfast00, 2023-03-30:
Right now, when the USB bridge feature is used, you need to manually assign an IP address to the device the ESP32 is plugged into: for example, if it is plugged into my laptop, I need to manually assign it 192.168.5.3. When I connect to the softap bridge network that is set up by the ESP32, I do get automatically assigned an IP address. Is it possible to also run a DHCP server on the USB interface?

Alternatively, in the case of using esp-mesh-lite, it would be nice if the USB interface appeared to be directly connected to the network the root node is connected to, so it could negotiate it's own ip address.

tswen, 2023-04-03:
Hi, You need to use IDF 5.0 version, as version 4.4 does not support multiple DHCPS.

redfast00, 2023-04-03:
Okay, thank you for the fast answer, will try that next.

closed 2023-04-03 by redfast00