espressif/esp-iot-bridge #41: esp_pbuf_allocate not found

gussstav, 2023-04-04:
I am trying to compile the 4G hotspot example, but I get this error message. I can't find esp_pbuf_allocate anywhere. Am I missing some include or component?

components\iot_bridge\src\bridge_spi.c:175:9: error: implicit declaration of function 'esp_pbuf_allocate' [-Werror=implicit-function-declaration]

tswen, 2023-04-04:
May I know what version of esp-idf you are using?

gussstav, 2023-04-04:
I am using 5.0.1, maybe I should downgrade to 5.0?

tswen, 2023-04-04:
Have you enabled the "BRIDGE_DATA_FORWARDING_NETIF_SPI" configuration option in menuconfig? I would like to know which netif you need to use for your project. 

I was able to compile using "release/v5.0", so it seems like there might be something wrong with your IDF environment or the submodules are not updated.

gussstav, 2023-04-04:
I did enable BRIDGE_DATA_FORWARDING_NETIF_SPI, I want to bridge an SPI ethernet controller with the 4G modem

I downgraded to V5.0 and reinstalled everything. It compiles now, thank you for your help