espressif/esp-iot-bridge #44: Ethernet and WiFi on same subnet

simon4096, 2023-04-18:
I have a project with one device connected to Ethernet port and several connected via WiFi.
Question: Would it be possible to have IP addresses for Ethernet and WIFI in the same subnet? Via DHCP?

tswen, 2023-04-18:
For the current architecture, since WiFi and Ethernet belong to two different interfaces, their network segments are also different. It is currently not possible to have two different types of devices in the same network segment.

simon4096, 2023-04-18:
Thank you for the quick answer.
Do you have any other ideas how I can achieve my goal? Since the WiFi clients may have multiple network cards, I would like to avoid having to set up a special route here.