espressif/esp-iot-bridge #45: More cellular module supports like Quectel EG25

dumtux, 2023-04-21:
Quectel released more different kinds of modems like EC20, EC25, and EG25.
Some of them support LTE and have better availability for mass production use.
Currently, I see only BG96 is supported by default.
How to make Quectel EG25 module work?
(Or is there any further plan to support more modules?)

tswen, 2023-04-23:
Actually, multiple types of modems are supported. You just need to modify the configuration items BRIDGE_MODEM_USB_VID, BRIDGE_MODEM_USB_PID, and BRIDGE_MODEM_USB_INTERFACE_NUMBER. If you only use the modem and SoftAP netif interfaces, it is recommended to try https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module.

dumtux, 2023-04-23:
That sounds good.
If I knew it, I should use USB to connect the modem.
However, I tried using UART just like the [TinyGSM](https://github.com/vshymanskyy/TinyGSM) Arduino library does, as It was my major lib to work with simple 4G module usage projects.
I'll connect ESP32-S3's USB to the modem in my next hardware revision.

Is there any way to make EG25/EC25 modules work with the UART connection?
Below is my connection diagram.

[image]

tswen, 2023-04-28:
Just use this [demo](https://github.com/espressif/esp-iot-bridge/tree/master/examples/4g_hotspot), then configure the tx, rx, and baud rate in menuconfig. It is recommended to adjust the baud rate of both the ESP32 and modem to 115200 for the first time.