espressif/esp-iot-bridge #47: Does esp32-C3 have a 4G to WiFi_AP routine?

JS-Ye, 2023-05-24:
### Does esp32-C3 have a 4G to WiFi_AP routine? Please help to provide. thank

### esp32-C3 有 4G转WiFi热点的例程吗？请帮忙提供一下。谢谢

tswen, 2023-05-24:
https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module
https://github.com/espressif/esp-iot-bridge/tree/master/examples/4g_hotspot
这两个都可以，如果只涉及到 4g modem 和 softap 这两个接口，推荐使用第一个

JS-Ye, 2023-05-24:
@tswen 好的，谢谢您！

JS-Ye, 2023-05-25:
@tswen 您好，现在这边用的是[4g_hotspot](https://github.com/espressif/esp-iot-bridge/tree/master/examples/4g_hotspot)例程，请问这个4G转WiFi 然后和4G串口连接硬件管脚只要两个rx tx就可以了，是吗？那些硬件流控制的需要不？

tswen, 2023-05-25:
还需要接 GND，低速情况下可以不需要流控

JS-Ye, 2023-05-26:
>目前模块发送esp好像收不到命令。然后试着接出串口，收到esp发送的命令后，串口功能回复命令，好像也是esp收不到。  

上面这个问题解决了，命令码回复错误了。  
现在是驱动4G模组后，然后就开始连接热点上网，显示没有网络，并报错误：esp-netif_lwip-ppp: pppos_input_topip failed with -1  
[image]
[image]

tswen, 2023-05-26:
115200 波特率速率太低了，可能会导致上网不通畅

tswen, 2023-05-30:
可以根据你们具体场景来调试，PCB 的话速率可以设置比较高，1-2Mbps 应该没问题，如果是杜邦线测试，太高的波特率可能会有问题。