espressif/esp-iot-bridge #48: Using the ETH interfce on the WT32-eth01

XLapAcme, 2023-05-29:
Hello, I'm trying to build a wifi to eth bridge using a WT32-eth01 module. I used the wifi-router example as a template. 

I managed to configure the BLE provisioning. I can connect to my local wifi just fine. I'm having trouble getting the ETH interface working. My local network is 192.168.5.0/24. For some reason, the eth interface get an ip in the 192.168.4.x range. I don't understand where that IP comes from. It's supposed to ask the DHCP server. I never see the status led on my network switch or the WT32 blinking so I'm under the impression the the ETH interface is simply not working. 

`I (0) cpu_start: Starting scheduler on APP CPU.
I (725) bridge_common: esp-iot-bridge version: 0.5.0
I (726) ip select: IP Address:192.168.4.1
I (726) ip select: GW Address:192.168.4.1
I (729) ip select: NM Address:255.255.255.0
I (734) system_api: Base MAC address is not set
I (739) system_api: read default base MAC address from EFUSE
I (745) mac select: MAC 0c:b8:15:4d:05:40
I (754) w5500.mac: version=0
I (765) esp_eth.netif.netif_glue: 02:00:00:12:34:56
I (765) esp_eth.netif.netif_glue: ethernet attached to netif
I (768) bridge_eth: Ethernet Started
E (773) w5500.mac: emac_w5500_transmit(473): free size (42) < send length (0)
Add netif eth with 2325731(commit id)
I (783) bridge_common: netif list add success
I (788) bridge_eth: ETH IP Address:192.168.4.1`

XLapAcme, 2023-05-29:
I'm new to using ESP board so just tell me if anymore infor is needed.

Here is the the sdkconfig.h I'm using: 
>#pragma once
#define CONFIG_SOC_BROWNOUT_RESET_SUPPORTED "Not determined"
#define CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED "Not determined"
#define CONFIG_SOC_DPORT_WORKAROUND "Not determined"
#define CONFIG_SOC_CAPS_ECO_VER_MAX 301
#define CONFIG_SOC_ADC_SUPPORTED 1
#define CONFIG_SOC_DAC_SUPPORTED 1
#define CONFIG_SOC_MCPWM_SUPPORTED 1
#define CONFIG_SOC_SDMMC_HOST_SUPPORTED 1
#define CONFIG_SOC_BT_SUPPORTED 1
#define CONFIG_SOC_PCNT_SUPPORTED 1
#define CONFIG_SOC_WIFI_SUPPORTED 1
#define CONFIG_SOC_SDIO_SLAVE_SUPPORTED 1
#define CONFIG_SOC_TWAI_SUPPORTED 1
#define CONFIG_SOC_EMAC_SUPPORTED 1
#define CONFIG_SOC_ULP_SUPPORTED 1
#define CONFIG_SOC_CCOMP_TIMER_SUPPORTED 1
#define CONFIG_SOC_RTC_FAST_MEM_SUPPORTED 1
#define CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED 1
#define CONFIG_SOC_RTC_MEM_SUPPORTED 1
#define CONFIG_SOC_I2S_SUPPORTED 1
#define CONFIG_SOC_RMT_SUPPORTED 1
#define CONFIG_SOC_SDM_SUPPORTED 1
#define CONFIG_SOC_SUPPORT_COEXISTENCE 1
#define CONFIG_SOC_AES_SUPPORTED 1
#define CONFIG_SOC_MPI_SUPPORTED 1
#define CONFIG_SOC_SHA_SUPPORTED 1
#define CONFIG_SOC_FLASH_ENC_SUPPORTED 1
#define CONFIG_SOC_SECURE_BOOT_SUPPORTED 1
#define CONFIG_SOC_TOUCH_SENSOR_SUPPORTED 1
#define CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL 5
#define CONFIG_SOC_XTAL_SUPPORT_26M 1
#define CONFIG_SOC_XTAL_SUPPORT_40M 1
#define CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT 1
#define CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED 1
#define CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED 1
#define CONFIG_SOC_ADC_DMA_SUPPORTED 1
#define CONFIG_SOC_ADC_PERIPH_NUM 2
#define CONFIG_SOC_ADC_MAX_CHANNEL_NUM 10
#define CONFIG_SOC_ADC_ATTEN_NUM 4
#define CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM 2
#define CONFIG_SOC_ADC_PATT_LEN_MAX 16
#define CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH 9
#define CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH 12
#define CONFIG_SOC_ADC_DIGI_RESULT_BYTES 2
#define CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV 4
#define CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH 2
#define CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW 20
#define CONFIG_SOC_ADC_RTC_MIN_BITWIDTH 9
#define CONFIG_SOC_ADC_RTC_MAX_BITWIDTH 12
#define CONFIG_SOC_RTC_SLOW_CLOCK_SUPPORT_8MD256 1
#define CONFIG_SOC_SHARED_IDCACHE_SUPPORTED 1
#define CONFIG_SOC_MMU_LINEAR_ADDRESS_REGION_NUM 5
#define CONFIG_SOC_CPU_CORES_NUM 2
#define CONFIG_SOC_CPU_INTR_NUM 32
#define CONFIG_SOC_CPU_HAS_FPU 1
#define CONFIG_SOC_CPU_BREAKPOINTS_NUM 2
#define CONFIG_SOC_CPU_WATCHPOINTS_NUM 2
#define CONFIG_SOC_CPU_WATCHPOINT_SIZE 64
#define CONFIG_SOC_DAC_PERIPH_NUM 2
#define CONFIG_SOC_DAC_RESOLUTION 8
#define CONFIG_SOC_GPIO_PORT 1
#define CONFIG_SOC_GPIO_PIN_COUNT 40
#define CONFIG_SOC_GPIO_VALID_GPIO_MASK 0xFFFFFFFFFF
#define CONFIG_SOC_GPIO_VALID_DIGITAL_IO_PAD_MASK 0xEF0FEA
#define CONFIG_SOC_I2C_NUM 2
#define CONFIG_SOC_I2C_FIFO_LEN 32
[... 613 lines cut ...]
#define CONFIG_BRIDGE_ETH_SPI_MOSI_GPIO 13
#define CONFIG_BRIDGE_ETH_SPI_MISO_GPIO 12
#define CONFIG_BRIDGE_ETH_SPI_CLOCK_MHZ 12
#define CONFIG_BRIDGE_ETH_SPI_CS0_GPIO 15
#define CONFIG_BRIDGE_ETH_SPI_INT0_GPIO 4
#define CONFIG_BRIDGE_ETH_SPI_PHY_RST0_GPIO -1
#define CONFIG_BRIDGE_ETH_SPI_PHY_ADDR0 1
#define CONFIG_BUTTON_PERIOD_TIME_MS 5
#define CONFIG_BUTTON_DEBOUNCE_TICKS 2
#define CONFIG_BUTTON_SHORT_PRESS_TIME_MS 180
#define CONFIG_BUTTON_LONG_PRESS_TIME_MS 1500
#define CONFIG_BUTTON_SERIAL_TIME_MS 20
#define CONFIG_ADC_BUTTON_MAX_CHANNEL 3
#define CONFIG_ADC_BUTTON_MAX_BUTTON_PER_CHANNEL 8
#define CONFIG_ADC_BUTTON_SAMPLE_TIMES 1
#define CONFIG_CU_DIAGNOSTICS_COLOR_ALWAYS 1
#define CONFIG_ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD 1
#define CONFIG_ESP_MODEM_CMUX_DELAY_AFTER_DLCI_SETUP 0
#define CONFIG_PROV_TRANSPORT_BLE 1
#define CONFIG_PROV_SECURITY_VERSION_2 1
#define CONFIG_PROV_SEC2_DEV_MODE 1
#define CONFIG_PROV_TRANSPORT 1
#define CONFIG_PROV_RESET_PROV_MGR_ON_FAILURE 1
#define CONFIG_PROV_MGR_MAX_RETRY_CNT 5
#define CONFIG_PROV_SHOW_QR 1

tswen, 2023-05-30:
You have selected the Station interface as the external netif, so it will obtain an IP address from the router as a DHCPC after connecting to it. Additionally, you have chosen the Ethernet interface as the data forwarding netif, so it will act as a DHCPS with the default IP subnet of [192.168.4.1](https://github.com/espressif/esp-iot-bridge/blob/master/components/iot_bridge/src/bridge_common.c#L149-L164). When an Ethernet device is connected to the ESP32, it will assign an IP address to the external device.

XLapAcme, 2023-05-30:
I see! 

I had hoped that the DHCP server from the wifi would assign an IP to the ETH interface too. I actually want my device on the wifi and the eth to be on the same subnet since my other eth connected device need a DHCP server to assign IP. I tought the bridge would just foward packet from both side transparantly. I guess it's not the case. 

Now that I think about it what I need is a layer 1 agnostic network switch. I don't even know if that's possible....

tswen, 2023-06-08:
Perhaps you can modify the [eth2ap](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/eth2ap) demo to achieve transparent data transmission between the STA and ETH interfaces.

XLapAcme, 2023-06-08:
I ended up buying a ready made eth/wifi bridge from Amazon for 30 buck instead. I don't think I have a deep enough understanding of ESP developpement to figure it out in a timely manner. Thanks anyway!

Also, for anyone reading this in the future, WT32-eth01 is not a great choice of board. information is really hard to find online and it isn't compatible with a bunch of library.

kiwihanshorst, 2023-09-16:
I found the subject discussed very interesting and am wondering if this is also an answer on what I want to do. Can someone please let me know if it is. 
Here is my situation:
I have an controller with an ethernet port, but do not have an ethernet connection locally available. I do however have wifi coverage. So my thought was to use the WT32-ETH01 module to connect to the available wifi and use the ethernet connection to connect my controller to the internet. Possible?

tswen, 2023-09-18:
Using the wifi_router example can solve your needs: https://github.com/espressif/esp-iot-bridge/tree/master/examples/wifi_router

kiwihanshorst, 2023-09-18:
Thanks, will have a look at it!

dheckl, 2024-07-23:
I try to get eth2ap running with WT32-ETH01, but I get "WiFi send packet failed: 12309" error. I already tried to set higher timeouts and queue size, but it didn't solved the issue. Has anyone got this running on the board?

tswen, 2024-07-24:
https://github.com/espressif/esp-idf/tree/master/examples/network/eth2ap#troubleshooting

dheckl, 2024-07-24:
I followed the troubleshooting steps and increased sending timeout as well queue size. Both have not changed anything.

dheckl, 2024-08-02:
eth2ap still not working on my side, but tried https://github.com/espressif/esp-idf/tree/release/v5.2/examples/network/bridge and it works perfectly.