espressif/esp-iot-bridge #50: After restarting WiFi_Router, it does not assign a fixed IP address to the client.

bluestar714, 2023-06-26:
I am using wifi_router example.
I have several client devices connected to wifi_router softap.
When I reboot wifi_router, the client devices are rebooted, but they seem to be assigned different IP addresses than before.

I have the following settings in sdkconfig, but it appears that when I turn off the wifi_router, the IP address configuration information is also lost.

Are there any other settings I need to make?

- sdkconfig
```
CONFIG_LWIP_DHCP_RESTORE_LAST_IP=y
```

esp-idf  is using nvs, so I think I'm storing data in a non-volatile place.

- dhcp_ip_addr_store.c

```
void dhcp_ip_addr_store(struct netif *netif)
{
    nvs_handle_t nvs;
    char if_key[IF_KEY_SIZE];
    if (netif == NULL) {
        return;
    }
    struct dhcp *dhcp = netif_dhcp_data(netif);
    uint32_t ip_addr = dhcp->offered_ip_addr.addr;

    if (nvs_open(DHCP_NAMESPACE, NVS_READWRITE, &nvs) == ESP_OK) {
        nvs_set_u32(nvs, gen_if_key(netif, if_key), ip_addr);
        nvs_commit(nvs);
        nvs_close(nvs);
    }
}

```

tswen, 2023-06-28:
Do you mean that "turning off the wifi_router" refers to restarting the wifi_router device?
If the wifi_router device is restarted, the protocol stack will not retain the previously assigned IP addresses for the corresponding devices.

bluestar714, 2023-06-28:
Thank you very much.

> Do you mean that "turning off the wifi_router" refers to restarting the wifi_router device?

Yes, it is.
I mean device reboot.
I was expecting a fixed IP address to be assigned to each client with a fixed lease time, similar to a typical WiFi router.

closed 2023-06-29 by bluestar714