espressif/esp-iot-bridge #53: warning: pointer 'item' used after 'free' [-Wuse-after-free]

rinstrum-franz, 2023-08-07:
```
/home/franz/git/rin/rinGate/rinGate/components/web_server/src/web_server.c: In function 'stop_scan_filter':
/home/franz/git/rin/rinGate/rinGate/components/web_server/src/web_server.c:290:103: warning: pointer 'fail_item' used after 'free' [-Wuse-after-free]
  290 |         for (router_obj_t *fail_item = SLIST_FIRST(&s_router_fail_list); fail_item != NULL; fail_item = SLIST_NEXT(fail_item, next)) {
      |                                                                                                       ^
/home/franz/git/rin/rinGate/rinGate/components/web_server/src/web_server.c:292:13: note: call to 'free' here
  292 |             free(fail_item);
      |             ^~~~~~~~~~~~~~~
/home/franz/git/rin/rinGate/rinGate/components/web_server/src/web_server.c: In function 'esp_web_start_scan_filter':
/home/franz/git/rin/rinGate/rinGate/components/web_server/src/web_server.c:612:73: warning: pointer 'item' used after 'free' [-Wuse-after-free]
  612 |         for (item = SLIST_FIRST(&s_router_all_list); item != NULL; item = SLIST_NEXT(item, next)) {
      |                                                                         ^
/home/franz/git/rin/rinGate/rinGate/components/web_server/src/web_server.c:615:13: note: call to 'free' here
  615 |             free(item);
      |             ^~~~~~~~~~
```

tswen, 2023-08-08:
Hello, thank you for your feedback. Although these warnings do not affect the functionality, we will work on optimizing and resolving these warnings in the future.

rinstrum-franz, 2023-08-08:
good