espressif/esp-iot-bridge #54: error: src/core/ipv4/ip4.c: Patch konnte nicht angewendet werden

rinstrum-franz, 2023-08-07:
Get this every time at Compile: 

```
-- Adding linker script /home/franz/esp-idf-v5.1/components/soc/esp32/ld/esp32.peripherals.ld
error: Anwendung des Patches fehlgeschlagen: src/core/ipv4/ip4.c:332
error: src/core/ipv4/ip4.c: Patch konnte nicht angewendet werden
```

Full log here: 

https://gist.github.com/rinstrum-franz/8fb27d45e1fab2b0428b50e74c913e71

tswen, 2023-08-08:
Hello, due to the fact that a part of the idf's lwip component hasn't been updated yet, we have created a patch. This patch will be applied to the lwip component within the iot_bridge component's CMake. If this patch has already been applied or if lwip has been modified in some other way, you might encounter the errors mentioned above. If you can ensure that the lwip patch has been applied, you can disregard the errors mentioned earlier.

```CMake
    execute_process(COMMAND
                    git apply ${CMAKE_CURRENT_LIST_DIR}/patch/ip4_forward.patch
                    WORKING_DIRECTORY $ENV{IDF_PATH}/components/lwip/lwip/)
```

rinstrum-franz, 2023-08-08:
understand.

Lennox7746, 2023-11-07:
请问idf5.1需要打ip4_forward.patch这个补丁吗？我在编译的时候报错error: patch failed: src/core/ipv4/ip4.c:332
error: src/core/ipv4/ip4.c: patch does not apply，按照你们的文档打补丁也报错error: patch failed: src/core/ipv4/ip4.c:332
error: src/core/ipv4/ip4.c: patch does not apply，请问下这个怎么能解决？

tswen, 2023-11-10:
目前是 CMake 自动给 idf 的 component/lwip/lwip 子模块打 patch 的，不需要用户手动打 patch。你可以去 esp-idf/components/lwip/lwip 里面看一下是否已经打上 patch 了，如果已经打上 patch 了，后面 CMake 编译会提示错误，但是可以忽略。

Talha-DPL, 2025-06-10:
Hello @tswen,

Description:
I'm encountering patch application failures when using the espressif/iot_bridge component (v1.0.1) with ESP-IDF v5.4.1. The following patches are failing to apply:

-- IOT_USBH_MODEM: .. error: patch failed: src/core/ipv4/dhcp.c:779 error: src/core/ipv4/dhcp.c: patch does not apply error: patch failed: components/lwip/apps/dhcpserver/dhcpserver.c:1267 error: components/lwip/apps/dhcpserver/dhcpserver.c: patch does not apply error: patch failed: src/core/ipv4/ip4.c:332 error: src/core/ipv4/ip4.c: patch does not apply error: patch failed: src/core/ipv4/ip4_napt.c:41 error: src/core/ipv4/ip4_napt.c: patch does not apply

Questions:
1. Are these patches still needed in ESP-IDF v5.4.1? The lwIP changelog shows several NAPT and DHCP improvements, but I need to confirm if these specific changes are included.
2. If the patches are still needed, how should they be updated for compatibility with ESP-IDF v5.4.1?
3. If the functionality is already included in v5.4.1, how can I verify this and safely remove these patches(there is no any code in these files related to patch)?

Environment:
- ESP-IDF Version: v5.4.1
- Component: espressif/iot_bridge v1.0.1
- Target: ESP32S3

Any guidance on verifying the functionality and resolving these patch conflicts would be greatly appreciated.

tswen, 2025-06-16:
https://github.com/espressif/esp-iot-bridge/tree/master/components/iot_bridge/patch

Here are the patch links that need to be applied. Before compilation, CMake will automatically patch the IDF. Normally, if the patches have already been applied to the IDF, subsequent builds will show patch does not apply - this is expected behavior. Alternatively, you can manually verify the patch content against the IDF modifications to ensure the patches were indeed applied.