espressif/esp-iot-bridge #55: Crash wen SPIRAM is on

franz-ms-muc, 2023-08-07:
See Log: 

https://gist.github.com/franz-ms-muc/8686894498e76fa004879489331b5a3b

tswen, 2023-08-08:
Hello, may I ask which version of the idf you are using? I recommend trying with idf version 5.0.

m5it, 2023-08-08:
me still on 4 didnt notice no problems. only that it leak of memory when x users are connected.. :)

tswen, 2023-08-09:
In general, whenever a child node connects to the hotspot of a parent node, the parent node will consume nearly 2k of memory. This is a necessary consumption on the Wi-Fi side, and I'm not sure if the `memory leak` you mentioned refers to this memory consumption.

franz-ms-muc, 2023-08-09:
> Hello, may I ask which version of the idf you are using? I recommend trying with idf version 5.0.

[**ESP-IDF Release v5.1**](https://github.com/espressif/esp-idf/releases/tag/v5.1)

**let me go for a test with** 

[**ESP-IDF Release v5.0.3**](https://github.com/espressif/esp-idf/releases/tag/v5.0.3)

franz-ms-muc, 2023-08-09:
so, result is that it is same. still crash when spiram on. 
i will test with 4.4.5 as well. ..

franz-ms-muc, 2023-08-09:
same in 4.4.5 
Crahsing.

franz-ms-muc, 2023-08-09:
there is still a Bug inside when using 4.4.5

see: 
https://github.com/franz-ms-muc/esp-iot-bridge/blob/ce247a249bc450469f42eb60fbd34b2d399e3800/main/app_main.c#L142
https://gist.github.com/franz-ms-muc/d999eb38eda77a0ab76bb2ca3220c935

after running the Sample all SPIRAM is gone, even if we have a lot before: 

before:
`MALLOC_CAP_: 8bit 4454202, 32bit 4488546, SPIRAM 4192135`
after: 
`MALLOC_CAP_: 8bit 99880, 32bit 134224, SPIRAM -247`

so something consumes away the SPIRAM. 
so i guess Problem always same, just the crash not happening in 4.4.5, because little bit under the Limit.

franz-ms-muc, 2023-08-09:
> In general, whenever a child node connects to the hotspot of a parent node, the parent node will consume nearly 2k of memory. This is a necessary consumption on the Wi-Fi side, and I'm not sure if the `memory leak` you mentioned refers to this memory consumption.

no. it is crashing without anybody has connected, directly at boot up.

tswen, 2023-08-14:
> there is still a Bug inside when using 4.4.5
>
> see: https://github.com/franz-ms-muc/esp-iot-bridge/blob/ce247a249bc450469f42eb60fbd34b2d399e3800/main/app_main.c#L142 [...]

I conducted testing using the branch you provided and found that it seems to be the esp_bridge_create_button API that is causing SPIRAM to become negative. Could you try disabling esp_bridge_create_button on your end?