espressif/esp-iot-bridge #56: esp_modem_set_mode failed

xiao-orien, 2023-08-09:
I use AIR724UG and ESP32-S3 with USB connection.
and config as:
CONFIG_BRIDGE_MODEM_USB_VID=0x1782
CONFIG_BRIDGE_MODEM_USB_PID=0x4E00
CONFIG_BRIDGE_MODEM_USB_INTERFACE_NUMBER=2

but report esp_modem_set_mode(ESP_MODEM_MODE_DATA) failed with -1:
I (0) cpu_start: App cpu up.
I (342) cpu_start: Pro cpu start user code
I (343) cpu_start: cpu freq: 160000000
I (343) cpu_start: Application information:
I (345) cpu_start: Project name:     4g_hotspot
I (351) cpu_start: App version:      v1.0.0-229-g95db7c1
I (356) cpu_start: Compile time:     Aug  9 2023 08:48:56
I (363) cpu_start: ELF file SHA256:  edc23e2f19407ba6...
I (369) cpu_start: ESP-IDF:          v4.4.4-dirty
I (374) heap_init: Initializing. RAM available for dynamic allocation:
I (381) heap_init: At 3FC9EAC0 len 0004AC50 (299 KiB): D/IRAM
I (388) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (394) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (400) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (407) spi_flash: detected chip: gd
I (411) spi_flash: flash io: dio
I (416) sleep: Configure to isolate all GPIO pins in sleep state
I (422) sleep: Enable automatic switching of GPIO sleep configuration
I (429) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (506) bridge_common: esp-iot-bridge version: 0.7.1
I (507) pp: pp rom version: e7ae62f
I (508) net80211: net80211 rom version: e7ae62f
I (512) wifi:wifi driver task: 3fceda94, prio:23, stack:6656, core=0
I (517) system_api: Base MAC address is not set
I (522) system_api: read default base MAC address from EFUSE
I (544) wifi:wifi firmware version: 6567a16
I (544) wifi:wifi certification version: v7.0
I (545) wifi:config NVS flash: enabled
I (545) wifi:config nano formating: disabled
I (549) wifi:Init data frame dynamic rx buffer num: 32
I (554) wifi:Init management frame dynamic rx buffer num: 32
I (559) wifi:Init management short buffer num: 32
I (563) wifi:Init dynamic tx buffer num: 32
I (567) wifi:Init static tx FG buffer num: 2
I (571) wifi:Init static rx buffer size: 1600
I (575) wifi:Init static rx buffer num: 10
I (579) wifi:Init dynamic rx buffer num: 32
I (584) wifi_init: rx ba win: 6
I (587) wifi_init: tcpip mbox: 32
I (591) wifi_init: udp mbox: 6
I (595) wifi_init: tcp mbox: 6
I (598) wifi_init: tcp tx win: 5744
I (603) wifi_init: tcp rx win: 5744
I (607) wifi_init: tcp mss: 1460
I (611) wifi_init: WiFi IRAM OP enabled
I (615) wifi_init: WiFi RX IRAM OP enabled
I (620) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11
I (664) wifi:mode : null
I (665) ip select: IP Address:192.168.4.1
I (665) ip select: GW Address:192.168.4.1
I (666) ip select: NM Address:255.255.255.0
I (670) bridge_wifi: IP Address:192.168.4.1
Add netif ap with 95db7c1(commit id)
I (678) bridge_common: netif list add success
I (683) wifi:mode : softAP (f4:12:fa:e2:0b:b9)
I (691) wifi:Total power save buffer number: 16
I (692) wifi:Init max length of beacon: 752/752
I (696) wifi:Init max length of beacon: 752/752
W (701) bridge_modem: Force reset 4g board
I (705) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (6214) bridge_modem: Initializing esp_modem for the BG96 module...
I (6265) bridge_modem: Waiting for USB device connection...
E (6667) HUB: LANGID 0x409 not found
E (6667) HUB: Stage failed: CHECK_FULL_LANGID_TABLE
I (7719) bridge_modem: Signal quality: rssi=99, ber=99
E (7806) bridge_modem: esp_modem_set_mode(ESP_MODEM_MODE_DATA) failed with -1
I (8784) wifi:Total power save buffer number: 16
I (8785) bridge_wifi: softap ssid: ESP_Bridge_e20bb9 password: 12345678

tswen, 2023-08-09:
If you intend to use a USB 4G module, I recommend using the following example. The USB host component that the iot-bridge project relies on doesn't provide full support for certain USB 4G modules yet.
https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module

xiao-orien, 2023-08-09:
But my esp-mesh-lite project must use 4G module access external network。In project config use 4G modem interface to connect to the external network, default use esp-iot-bridge.