espressif/esp-iot-bridge #57: Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

xiao-orien, 2023-08-09:
I use AIR780E and ESP32-S3 with USB
config：
CONFIG_BRIDGE_MODEM_USB_VID=0x19D1
CONFIG_BRIDGE_MODEM_USB_PID=0x1
CONFIG_BRIDGE_MODEM_USB_INTERFACE_NUMBER=2

Same issues as closed issues:
Configuration descriptor larger than control transfer max length AIR780E and ESP32-S3 with USB connection (AEGHB-88)

Then,
change the configuration item CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE to 1024.

But,got the other error:
I (291) cpu_start: App cpu up.
I (341) cpu_start: Pro cpu start user code
I (341) cpu_start: cpu freq: 160000000
I (341) cpu_start: Application information:
I (344) cpu_start: Project name:     4g_hotspot
I (349) cpu_start: App version:      v1.0.0-229-g95db7c1
I (355) cpu_start: Compile time:     Aug  9 2023 10:02:06
I (361) cpu_start: ELF file SHA256:  6a3f544199d54c13...
I (367) cpu_start: ESP-IDF:          v4.4.4-dirty
I (372) heap_init: Initializing. RAM available for dynamic allocation:
I (380) heap_init: At 3FC9EA50 len 0004ACC0 (299 KiB): D/IRAM
I (386) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (393) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (399) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (406) spi_flash: detected chip: gd
I (410) spi_flash: flash io: dio
I (414) sleep: Configure to isolate all GPIO pins in sleep state
I (420) sleep: Enable automatic switching of GPIO sleep configuration
I (427) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (504) bridge_common: esp-iot-bridge version: 0.7.1
I (505) pp: pp rom version: e7ae62f
I (506) net80211: net80211 rom version: e7ae62f
I (510) wifi:wifi driver task: 3fceda94, prio:23, stack:6656, core=0
I (515) system_api: Base MAC address is not set
I (520) system_api: read default base MAC address from EFUSE
I (542) wifi:wifi firmware version: 6567a16
I (542) wifi:wifi certification version: v7.0
I (542) wifi:config NVS flash: enabled
I (543) wifi:config nano formating: disabled
I (547) wifi:Init data frame dynamic rx buffer num: 32
I (551) wifi:Init management frame dynamic rx buffer num: 32
I (557) wifi:Init management short buffer num: 32
I (561) wifi:Init dynamic tx buffer num: 32
I (565) wifi:Init static tx FG buffer num: 2
I (569) wifi:Init static rx buffer size: 1600
I (573) wifi:Init static rx buffer num: 10
I (577) wifi:Init dynamic rx buffer num: 32
I (581) wifi_init: rx ba win: 6
I (585) wifi_init: tcpip mbox: 32
I (589) wifi_init: udp mbox: 6
I (592) wifi_init: tcp mbox: 6
I (596) wifi_init: tcp tx win: 5744
I (600) wifi_init: tcp rx win: 5744
I (604) wifi_init: tcp mss: 1460
I (608) wifi_init: WiFi IRAM OP enabled
I (613) wifi_init: WiFi RX IRAM OP enabled
I (618) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11
I (665) wifi:mode : null
I (667) ip select: IP Address:192.168.4.1
I (667) ip select: GW Address:192.168.4.1
I (667) ip select: NM Address:255.255.255.0
I (671) bridge_wifi: IP Address:192.168.4.1
Add netif ap with 95db7c1(commit id)
I (679) bridge_common: netif list add success
I (685) wifi:mode : softAP (f4:12:fa:e2:0b:b9)
I (693) wifi:Total power save buffer number: 16
[... 68 lines cut ...]
I (57) boot.esp32s3: Boot SPI Speed : 80MHz
I (62) boot.esp32s3: SPI Mode       : DIO
I (66) boot.esp32s3: SPI Flash Size : 16MB
I (71) boot: Enabling RNG early entropy source...
I (77) boot: Partition Table:
I (80) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (95) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (102) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (110) boot:  3 factory          factory app      00 00 00010000 00100000
I (118) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (125) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (133) boot: End of partition table
I (137) boot: Defaulting to factory image
I (142) boot_comm: chip revision: 1, min. application chip revision: 0
I (149) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=1bef0h (114416) map
I (178) esp_image: segment 1: paddr=0002bf18 vaddr=3fc96d80 size=04100h ( 16640) load
I (182) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=89bb4h (564148) map
I (285) esp_image: segment 3: paddr=000b9bdc vaddr=3fc9ae80 size=003b8h (   952) load
I (285) esp_image: segment 4: paddr=000b9f9c vaddr=40374000 size=12d78h ( 77176) load
I (315) boot: Loaded app from partition at offset 0x10000
I (315) boot: Disabling RNG early entropy source...
I (327) cpu_start: Pro cpu up.
I (327) cpu_start: Starting app cpu, entry point is 0x403753b0
0x403753b0: call_start_cpu1 at D:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_system/port/cpu_start.c:148

tswen, 2023-08-09:
If you intend to use a USB 4G module, I recommend using the following example. The USB host component that the iot-bridge project relies on doesn't provide full support for certain USB 4G modules yet.
https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module

Beardsome, 2025-05-12:
select correct ESP32-S3 version in SDK config it will solve your issue