espressif/esp-iot-bridge #58: Is it possible to make the SoftAP IP range have same IP with the IP that Wi-Fi STA connected to?

anhnt-dev, 2023-09-09:
Hi,
as the subject question, Is it possible? Currently, SoftAP will start a DHCP server to provide IP to the client that is connected to, and STA will get IP from Router. It will make two different networks. The SoftAP's client can ping to any device on the Router's network, but devices from the Router's network can't talk to SoftAp's client without port forwarding.

anhnt-dev, 2023-09-18:
> Hello, for the IoT-Bridge architecture, since it involves NAPT, the IP subnet for SoftAP must be distinct from the IP subnet for the station. However, if you're referring to something like the ESP32 establishing a transparent link between the router and SoftAP, you can implement it by modifying the example for creating a transparent link between Ethernet (eth) and SoftAP. Here's the link to the example:
> 
> https://github.com/espressif/esp-idf/tree/master/examples/network/eth2ap

Thanks @tswen, but is it possible for us to forward the packet between wifi Softap & STA directly? like eth2ap.

tswen, 2023-09-21:
Yes, it is possible