espressif/esp-iot-bridge #62: 过几个小时再连接ap时 没有网络

qq547176052, 2023-12-27:
[image]
使用的是 wired_nic 项目
打开ap后手机连接是有网络的
过一段时间后再来连接就没有网络了
下面是代码

```app_main.c
/*
 * SPDX-FileCopyrightText: 2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <string.h>
#include "nvs.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "esp_event.h"
#include "esp_bridge.h"
#include "esp_wifi.h"

static esp_err_t esp_storage_init(void)
{
    esp_err_t ret = nvs_flash_init();

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }

    return ret;
}

void app_main(void)
{
    esp_storage_init();

    /* Init and register system/core components */
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_bridge_create_all_netif();

    wifi_config_t wifi_cfg = {
        .ap = {
            .ssid = "111",
            .password = "12345678",
        }
    };
    esp_bridge_wifi_set_config(WIFI_IF_AP, &wifi_cfg);
}
```

tswen, 2023-12-28:
能否在示例里面增加一个定时 ping 的操作，如果以太网 netif 能够正常访问互联网，softap 理论上来说也是没问题的。

tswen, 2024-01-02:
idf 里面有个 dhcpserver.c 文件，可以将 DHCPS_DEBUG 宏打开之后看一下，如果能抓空中包，可以配合抓包看下。

qq547176052, 2024-01-03:
不会抓包,不用几个小时,大概10分钟左右 就没网络了,硬件换过2家设备,测试结果都一样
[image]

[image]
[image]

qq547176052, 2024-01-12:
[image]
打开debug后输出这些

tswen, 2024-01-17:
听您的描述是分配 ip 出了问题，那么您应该将 dhcps 的 debug 打开（https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c#L66）。

qq547176052, 2024-01-29:
打开了调试 没有看到有用的消息
[image]

qq547176052, 2024-01-29:
[wired_nic.zip](https://github.com/espressif/esp-iot-bridge/files/14085999/wired_nic.zip)
这是我的项目 你能帮我看下是不是配置的问题吗

tswen, 2024-03-03:
可以打印下内存确认下是否是内存不足了，
```c
        printf("free_heap_size:%d\r\n", esp_get_free_heap_size());
        printf("free_internal_heap_size:%d\r\n", esp_get_free_internal_heap_size());
        printf("minimum_free_heap_size:%d\r\n\r\n", esp_get_minimum_free_heap_size());
```