espressif/esp-iot-bridge #63: 是否同時使用多個SPI外設會造成esp-iot-bridge崩潰?

8888alex, 2024-01-11:
### 說明
不好意思，不知道這問題發在這裡合不合適。若不合適還請告知要發在哪裡，謝謝。
我需要使用 [MB85RS16](https://datasheet.lcsc.com/lcsc/1811031915_FUJITSU-MB85RS16PNF-G-JNERE1_C38647.pdf) 及 [RS522 RFID MODULE](https://www.makerskart.com/shop/arduino/arduino-acc/rs522-rfid-module/) 這兩個外設，
同時還有 mesh-lite(其中會調用espressif__iot_bridge)。
當工程中同時調用這三者中的任兩個都可以正常運行，但同時三個時便會崩潰。
也之所以會發在這裡是因為前面兩個SPI外設都有正常運行並讀取資料
這段表示FRAM正常讀寫
```
spi_fram_read_uint32(305419896)
out_num: 305419896
I (1154) Fram: has fram
```
這段表示RC522正常讀寫
```
I (2744) rfid_rc522: Tag scanned (sn: 418046553)
```
最後查到的線索是在 iot_bridge 調用的過程中崩潰。
請問我還能做些甚麼檢查?

### 環境:
esp32-s2-devkitm-1-v1
IDF v5.0.2

idf_component.yml
dependencies:
    idf: ">=5.0"
    mesh_lite:
        version: "*"
    usb_device:
        path: components/usb/usb_device
        git: https://github.com/espressif/esp-iot-bridge.git
        rules:
        - if: "target in [esp32s2, esp32s3]"
    espressif/led_strip: "^2.0.0"
    espressif/esp_encrypted_img: "^2.2.0"

### LOG
先調用FRAM及RC522後再調用iot-bridge所產生生的錯誤日誌
[LOG_iot_bridge.txt](https://github.com/espressif/esp-iot-bridge/files/13903691/LOG_iot_bridge.txt)
下面作為對照，先調用FRAM及iot-bridg後再調用RC522所產生生的錯誤日誌
[LOG_RC522.txt](https://github.com/espressif/esp-iot-bridge/files/13903758/LOG_RC522.txt)

tswen, 2024-01-12:
感觉像是 RC522 的使用导致了内存踩踏之类的一些问题。
第一份日志，RC522 起来之后直接影响了 wifi 的初始化（0x400bc70e: esp_wifi_init at C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_wifi/src/wifi_init.c:249），跟 iot_bridge 没有关系。
第二份日志，在没有启用 RC522 之前，wifi 可以正常初始化，但是 RC522 起来之后，破坏了 wifi 底层 ap_rx_cb，导致错误。

请仔细检查下 RC522 相关的代码，确保没有越界访问数组、释放已释放的内存或者其他可能导致内存访问错误的问题。

8888alex, 2024-01-15:
經測試問題又回到了 PSRAM上面了
我用您在[issuecomment-1874823433](https://github.com/espressif/esp-mesh-lite/issues/51#issuecomment-1874823433)所提供的範例中添加fram跟rc522相關的代碼，在不開啟 Support for external, SPI-connected RAM設置的情況下是可以正常運行。
[LOG_正常運行.txt](https://github.com/espressif/esp-iot-bridge/files/13933662/LOG_.txt)

但我的工程需要更多的RAM所需要開啟。但開啟後便會導致崩潰
[LOG_崩潰.txt](https://github.com/espressif/esp-iot-bridge/files/13933693/LOG_.txt)
[樣本代碼](https://github.com/espressif/esp-iot-bridge/files/13933683/ssl.zip)

請問我該如何配置PSRAM以得到更多的RAM並正常運行?
或是我該將這個問題發到哪個ISSUES?
謝謝您的時間。

tswen, 2024-01-15:
https://docs.espressif.com/projects/esp-idf/zh_CN/v5.1.2/esp32/api-guides/external-ram.html
可以参考下这个链接，不确定会不会对您有些帮助。

8888alex, 2024-01-16:
好的我研究一下，
感謝您的時間。

closed 2024-01-16 by 8888alex