espressif/esp-iot-bridge #67: Failed to build the wifi_router example

PabasaraDilshan, 2024-02-08:
failed to build the wifi_router example
```
Executing action: all (aliases: build)
Running ninja in directory /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32
NOTICE: Skipping optional dependency: usb_device
Manifest files have changed, solving dependencies.
Using component placed at /Users/dilsh/Documents/fyp/esp-iot-bridge/components/iot_bridge for dependency espressif/iot_bridge(*), specified in /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/main/idf_component.yml
....NOTICE: Skipping optional dependency: espressif/esp_modem_usb_dte
........Updating lock file at /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/dependencies.lock
Processing 8 dependencies:
[8/8] wifi_prov_mgr (d98493557e0a21e4fc17bd42402520fbd1941c17)
NOTICE: Skipping optional dependency: espressif/esp_modem_usb_dte
NOTICE: Skipping optional dependency: usb_device
-- Project sdkconfig file /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/sdkconfig
warning: the int symbol BRIDGE_SOFTAP_MAX_CONNECT_NUMBER (defined at /Users/dilsh/Documents/fyp/esp-iot-bridge/components/iot_bridge/Kconfig:90) has a non-int default BRIDGE_CUSTOM_MAX_CONNECT_NUMBER (undefined)
Loading defaults file /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/sdkconfig.defaults...
warning: unknown kconfig symbol 'MEMMAP_SMP' assigned to 'y' in /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/sdkconfig.defaults
warning: unknown kconfig symbol 'MODEM_LEGACY_API' assigned to 'n' in /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/sdkconfig.defaults
Loading defaults file /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/sdkconfig.defaults.esp32...
warning: unknown kconfig symbol 'MEMMAP_SMP' assigned to 'y' in /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'MODEM_LEGACY_API' assigned to 'n' in /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/sdkconfig.defaults.esp32
-- Compiler supported targets: xtensa-esp-elf
-- App "wifi_router" version: v1.0.0-255-gd984935
-- Adding linker script /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/dilsh/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /Users/dilsh/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/dilsh/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /Users/dilsh/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/dilsh/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/dilsh/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/dilsh/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Users/dilsh/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- BUTTON: 2.5.6
error: patch failed: src/core/ipv4/ip4.c:332
error: src/core/ipv4/ip4.c: patch does not apply
-- IOT_BRIDGE: 0.11.0
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_driver_ana_cmpr esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__button espressif__cmake_utilities espressif__esp_modem espressif__qrcode esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 iot_bridge json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling web_server wifi_prov_mgr wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/dilsh/esp/esp-idf/components/app_trace /Users/dilsh/esp/esp-idf/components/app_update /Users/dilsh/esp/esp-idf/components/bootloader /Users/dilsh/esp/esp-idf/components/bootloader_support /Users/dilsh/esp/esp-idf/components/bt /Users/dilsh/esp/esp-idf/components/cmock /Users/dilsh/esp/esp-idf/components/console /Users/dilsh/esp/esp-idf/components/cxx /Users/dilsh/esp/esp-idf/components/driver /Users/dilsh/esp/esp-idf/components/efuse /Users/dilsh/esp/esp-idf/components/esp-tls /Users/dilsh/esp/esp-idf/components/esp_adc /Users/dilsh/esp/esp-idf/components/esp_app_format /Users/dilsh/esp/esp-idf/components/esp_bootloader_format /Users/dilsh/esp/esp-idf/components/esp_coex /Users/dilsh/esp/esp-idf/components/esp_common /Users/dilsh/esp/esp-idf/components/esp_driver_ana_cmpr /Users/dilsh/esp/esp-idf/components/esp_driver_dac /Users/dilsh/esp/esp-idf/components/esp_driver_gpio /Users/dilsh/esp/esp-idf/components/esp_driver_gptimer /Users/dilsh/esp/esp-idf/components/esp_driver_i2c /Users/dilsh/esp/esp-idf/components/esp_driver_i2s /Users/dilsh/esp/esp-idf/components/esp_driver_ledc /Users/dilsh/esp/esp-idf/components/esp_driver_mcpwm /Users/dilsh/esp/esp-idf/components/esp_driver_parlio /Users/dilsh/esp/esp-idf/components/esp_driver_pcnt /Users/dilsh/esp/esp-idf/components/esp_driver_rmt /Users/dilsh/esp/esp-idf/components/esp_driver_sdio /Users/dilsh/esp/esp-idf/components/esp_driver_sdm /Users/dilsh/esp/esp-idf/components/esp_driver_sdmmc /Users/dilsh/esp/esp-idf/components/esp_driver_sdspi /Users/dilsh/esp/esp-idf/components/esp_driver_spi /Users/dilsh/esp/esp-idf/components/esp_driver_tsens /Users/dilsh/esp/esp-idf/components/esp_driver_uart /Users/dilsh/esp/esp-idf/components/esp_driver_usb_serial_jtag /Users/dilsh/esp/esp-idf/components/esp_eth /Users/dilsh/esp/esp-idf/components/esp_event /Users/dilsh/esp/esp-idf/components/esp_gdbstub /Users/dilsh/esp/esp-idf/components/esp_hid /Users/dilsh/esp/esp-idf/components/esp_http_client /Users/dilsh/esp/esp-idf/components/esp_http_server /Users/dilsh/esp/esp-idf/components/esp_https_ota /Users/dilsh/esp/esp-idf/components/esp_https_server /Users/dilsh/esp/esp-idf/components/esp_hw_support /Users/dilsh/esp/esp-idf/components/esp_lcd /Users/dilsh/esp/esp-idf/components/esp_local_ctrl /Users/dilsh/esp/esp-idf/components/esp_mm /Users/dilsh/esp/esp-idf/components/esp_netif /Users/dilsh/esp/esp-idf/components/esp_netif_stack /Users/dilsh/esp/esp-idf/components/esp_partition /Users/dilsh/esp/esp-idf/components/esp_phy /Users/dilsh/esp/esp-idf/components/esp_pm /Users/dilsh/esp/esp-idf/components/esp_psram /Users/dilsh/esp/esp-idf/components/esp_ringbuf /Users/dilsh/esp/esp-idf/components/esp_rom /Users/dilsh/esp/esp-idf/components/esp_system /Users/dilsh/esp/esp-idf/components/esp_timer /Users/dilsh/esp/esp-idf/components/esp_wifi /Users/dilsh/esp/esp-idf/components/espcoredump /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/managed_components/espressif__button /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/managed_components/espressif__cmake_utilities /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/managed_components/espressif__esp_modem /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/managed_components/espressif__qrcode /Users/dilsh/esp/esp-idf/components/esptool_py /Users/dilsh/esp/esp-idf/components/fatfs /Users/dilsh/esp/esp-idf/components/freertos /Users/dilsh/esp/esp-idf/components/hal /Users/dilsh/esp/esp-idf/components/heap /Users/dilsh/esp/esp-idf/components/http_parser /Users/dilsh/esp/esp-idf/components/idf_test /Users/dilsh/esp/esp-idf/components/ieee802154 /Users/dilsh/Documents/fyp/esp-iot-bridge/components/iot_bridge /Users/dilsh/esp/esp-idf/components/json /Users/dilsh/esp/esp-idf/components/log /Users/dilsh/esp/esp-idf/components/lwip /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/main /Users/dilsh/esp/esp-idf/components/mbedtls /Users/dilsh/esp/esp-idf/components/mqtt /Users/dilsh/esp/esp-idf/components/ne [... cut ...]
/Users/dilsh/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/dilsh/esp/esp-idf/components/esp_coex/lib/esp32/libcoexist.a(coexist_hw.o):(.iram1.17+0x16): undefined reference to `btdm_rf_bb_reg_init'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/build/log/idf_py_stderr_output_37137 and /Users/dilsh/Documents/fyp/esp-iot-bridge/examples/wifi_router/build/log/idf_py_stdout_output_37137
```

PabasaraDilshan, 2024-02-13:
Even if i write a my own wifi router code to a esp32, connected deviceses can't have the internet access? Can any one please clarify is this library working or not

tswen, 2024-02-17:
What is your idf version? Is there something wrong with idf, or is the esp-iot-bridge repository up to date?