espressif/esp-iot-bridge #68: Special character &

miccgn, 2024-02-24:
Hi,

from my experience, the code is not able to deal with the &-sign, if this is included in either the SSID or the STA password.

In both cases, the content gets truncated before the &-sign when stored to NVS.

tswen, 2024-03-03:
Hello, thank you for your feedback. We will test and verify this part.

tswen, 2024-03-05:
[image]
Hello, I have conducted a test, and NVS does not truncate the "&" symbol.

miccgn, 2024-03-05:
Hi @tswen,

thanks for your analysis. However, when I tried the example "wifi_router", I had troubles connecting to an SSID with & characters. 

I can't reproduce it at the moment, as my only ESP32 is in use now, so let's close this here. For some reason (output messages on the webpage?) I assumed there was an error in the parsing routines.

closed 2024-03-05 by miccgn