espressif/esp-iot-bridge #69: Bridge (Host USB Vendor Specific to Wifi) - Multi hubs and multi devices

Docfch, 2024-03-02:
Hello Everyone,
I am currently working on a project where I use a USB cable for communication between my stm32 in vendor-specific device mode and a PC software with libusb, and I want to replace it with a connection of my board to the USB of the esp32 s3 to communicate through WiFi.
For this, I wanted to know when will we have a bridge (host USB vendor specific to wifi) example ?
When multi hubs and multi devices connected to a host will be supported ?
Thanks.

tswen, 2024-03-03:
If USB RNDIS or USB ECM is required, you can use the wired_nic or wireless_nic examples. You need to manually enable the CONFIG_BRIDGE_FORWARDING_NETIF_USB configuration option in menuconfig.

Docfch, 2024-03-03:
Does it work with multi hubs and multi devices ?

tswen, 2024-03-04:
The device should be directly plugged into the USB host device. Plugging it into a hub may not work.