espressif/esp-iot-bridge #72: Unable to build binary due to ip_napt_enable function

BenjaminDeHeisenberg, 2024-04-25:
Recreating the problem:
IDF docker Image used: v4.4.7
compiling the code to use esp32 as the 4g hotspot

1. downloaded the git repository
2. went to the esp-iot-bridge/test_app 
3. ran idf.menuconfig

Changes made in menuconfig: 
Component config -> Bridge configuration -> The interface used for connecting the external network : 4G modem
Component config -> Bridge configuration -> The interface used to provide network data... : use Wifi SoftAP
SoftAP Config-> default values

Modem setting: changed to SIM7600

4. ran idf.py build

Problem when executing idf.py 

```
[1349/1375] Building C object esp-idf/iot_bridge/CMakeFiles/__idf_iot_bridge.dir/src/bridge_common.c.obj
FAILED: esp-idf/iot_bridge/CMakeFiles/__idf_iot_bridge.dir/src/bridge_common.c.obj
...
/project/esp-iot-bridge/components/iot_bridge/src/bridge_common.c: In function 'esp_bridge_netif_set_ip_info':
/project/esp-iot-bridge/components/iot_bridge/src/bridge_common.c:487:13: error: implicit declaration of function 'ip_napt_enable'; did you mean 'xthal_int_enable'? [-Werror=implicit-function-declaration]
             ip_napt_enable(ip_info->ip.addr, 1);
             ^~~~~~~~~~~~~~
             xthal_int_enable
cc1: some warnings being treated as errors
[1358/1375] Building CXX object esp-idf/espressif__esp_modem/CMakeFiles/__idf_espressif__esp_modem.dir/src/esp_modem_command_library.cpp.obj 
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Things that I have tried out: 
1. Build using the latest docker idf image: idf version 5.3, 
4. build using idf version 4.4.7
5. build using idf version 4.3.3

tswen, 2024-04-26:
Is it that you haven't enabled the following two configuration options?

CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y

BenjaminDeHeisenberg, 2024-04-26:
I made the suggested changes in the sdkconfig file and that gave a new error: 

```
 /project/esp-iot-bridge/components/iot_bridge/src/bridge_modem.c: In function 'esp_bridge_create_modem_netif':
/project/esp-iot-bridge/components/iot_bridge/src/bridge_modem.c:130:43: error: implicit declaration of function 'ESP_NETIF_DEFAULT_PPP'; did you mean 'ESP_NETIF_DEFAULT_ETH'? [-Werror=implicit-function-declaration]
  130 |     esp_netif_config_t netif_ppp_config = ESP_NETIF_DEFAULT_PPP();
      |                                           ^~~~~~~~~~~~~~~~~~~~~
      |                                           ESP_NETIF_DEFAULT_ETH
/project/esp-iot-bridge/components/iot_bridge/src/bridge_modem.c:130:43: error: invalid initializer
cc1: some warnings being treated as errors
[1253/1263] Linking C static library esp-idf/espressif__esp_modem/libespressif__esp_modem.a
ninja: build stopped: subcommand failed.
```

tswen, 2024-04-26:
# Override some defaults to enable PPP
CONFIG_LWIP_PPP_SUPPORT=y
CONFIG_LWIP_PPP_PAP_SUPPORT=y

BenjaminDeHeisenberg, 2024-04-26:
I was able to successfully build the test_app and flash the code onto the esp32. but as suggested, it is for carrying out tests. 
I want to make a simple 4G-Hotspot example implemented on an ESP32 wroom32 that is connected to a SIM7600 module 

in the examples/4G_hotspot folder,I am getting an error while executing the `idf.py menuconfig `command

This is the error: 
```
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
NOTICE: Skipping optional dependency: usb_device
Using component placed at /projects/esp-iot-bridge/components/iot_bridge for dependency espressif/iot_bridge(*), specified in /projects/esp-iot-bridge/examples/4g_hotspot/main/idf_component.yml
....HINT: Component "espressif/usb_host_cdc_acm" has suitable versions for other targets: "esp32s3", "esp32s2", "esp32p4". Is your current target "esp32" set correctly?
...HINT: Component "espressif/usb_host_cdc_acm" has suitable versions for other targets: "esp32s3", "esp32s2". Is your current target "esp32" set correctly?
...CMake Error at /opt/esp/idf/tools/cmake/build.cmake:463 (message):
  WARNING: Component "espressif/usb_host_cdc_acm" not found

  WARNING: Component "espressif/usb_host_cdc_acm" not found

  WARNING: Component "espressif/usb_host_cdc_acm" not found

  ERROR: Because no versions of espressif/esp_modem_usb_dte match
  >1.0.0,<1.1.0 || >1.1.0,<1.1.0~1 || >1.1.0~1,<2.0.0

   and espressif/esp_modem_usb_dte (1.0.0) depends on espressif/esp_modem (^0.1.23), espressif/esp_modem_usb_dte (>=1.0.0,<1.1.0 || >1.1.0,<1.1.0~1 || >1.1.0~1,<2.0.0) requires espressif/esp_modem (^0.1.23).

  And because espressif/esp_modem_usb_dte (1.1.0) depends on
  espressif/usb_host_cdc_acm (2.*), espressif/esp_modem_usb_dte
  (>=1.0.0,<1.1.0~1 || >1.1.0~1,<2.0.0) requires espressif/esp_modem
  (^0.1.23) or espressif/usb_host_cdc_acm (2.*).

  And because espressif/iot_bridge (0.11.3) depends on both
  espressif/esp_modem (1.*) and espressif/esp_modem_usb_dte (1.*),
  espressif/iot_bridge (0.11.3) requires espressif/usb_host_cdc_acm (2.*) or
  espressif/esp_modem_usb_dte (1.1.0~1).

  And because no versions of espressif/iot_bridge match >=0.0.0,<0.11.3 ||
  >0.11.3

   and espressif/esp_modem_usb_dte (1.1.0~1) depends on espressif/usb_host_cdc_acm (2.*), espressif/iot_bridge (>=0.0.0) requires espressif/usb_host_cdc_acm (>=2.0.0,<3.0.0).

  So, because no versions of espressif/usb_host_cdc_acm match 2.*

   and project depends on espressif/iot_bridge (*), version solving failed.

Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/projects/esp-iot-bridge/examples/4g_hotspot/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

BenjaminDeHeisenberg, 2024-04-27:
I was able to build The 4G hotspot firmware with the v1.0.0 release of this repository. 
I am currently facing another issue.

Problem:
Unable to download the OTA on another ESP32. 

Recreation of the issue:
I am currently using two ESP32 for my project. The first ESP 32 is flashed with the 4G hotspot example and it provides Wi-Fi connectivity to the second ESP 32. The second ESP 32 uses Blynk and is used for other operations. 
The Wi-Fi connectivity to the second ESP 32 is stable and it is capable to communicate with the blink server. But it is unable to perform an OTA firmware  update. 
When the second ESP32 is connected to the local Wi-Fi network or to a mobile hotspot It is capable of performing an OTA without any issue but when it is connected to the first ESP 32 it is not able to perform an OTA.

tswen, 2024-04-28:
> I was able to successfully build the test_app and flash the code onto the esp32. but as suggested, it is for carrying out tests. I want to make a simple 4G-Hotspot example implemented on an ESP32 [...]

您可以将 idf-component-manager 回退到 1.4.1 试一下
`pip install idf-component-manager==1.4.1`

benjamin-sl, 2024-05-03:
> > I was able to successfully build the test_app and flash the code onto the esp32. but as suggested, it is for carrying out tests. I want to make a simple 4G-Hotspot example implemented on an ESP32 wroom32 that is connected to a SIM7600 module
> > in the examples/4G_hotspot folder,I am getting an error while executing the `idf.py menuconfig `command
> > This is the error:
> > ```
> > -- Detecting CXX compile features
> > -- Detecting CXX compile features - done
> > -- Building ESP-IDF components for target esp32
> > Dependencies lock doesn't exist, solving dependencies.
> > NOTICE: Skipping optional dependency: usb_device
> > Using component placed at /projects/esp-iot-bridge/components/iot_bridge for dependency espressif/iot_bridge(*), specified in /projects/esp-iot-bridge/examples/4g_hotspot/main/idf_component.yml
> > ....HINT: Component "espressif/usb_host_cdc_acm" has suitable versions for other targets: "esp32s3", "esp32s2", "esp32p4". Is your current target "esp32" set correctly?
> > ...HINT: Component "espressif/usb_host_cdc_acm" has suitable versions for other targets: "esp32s3", "esp32s2". Is your current target "esp32" set correctly?
> > ...CMake Error at /opt/esp/idf/tools/cmake/build.cmake:463 (message):
> >   WARNING: Component "espressif/usb_host_cdc_acm" not found
> > 
> >   WARNING: Component "espressif/usb_host_cdc_acm" not found
> > 
> >   WARNING: Component "espressif/usb_host_cdc_acm" not found
> > 
> >   ERROR: Because no versions of espressif/esp_modem_usb_dte match
> >   >1.0.0,<1.1.0 || >1.1.0,<1.1.0~1 || >1.1.0~1,<2.0.0
> > 
> >    and espressif/esp_modem_usb_dte (1.0.0) depends on espressif/esp_modem (^0.1.23), espressif/esp_modem_usb_dte (>=1.0.0,<1.1.0 || >1.1.0,<1.1.0~1 || >1.1.0~1,<2.0.0) requires espressif/esp_modem (^0.1.23).
> > 
> >   And because espressif/esp_modem_usb_dte (1.1.0) depends on
> >   espressif/usb_host_cdc_acm (2.*), espressif/esp_modem_usb_dte
> >   (>=1.0.0,<1.1.0~1 || >1.1.0~1,<2.0.0) requires espressif/esp_modem
> >   (^0.1.23) or espressif/usb_host_cdc_acm (2.*).
> > 
> >   And because espressif/iot_bridge (0.11.3) depends on both
> >   espressif/esp_modem (1.*) and espressif/esp_modem_usb_dte (1.*),
> >   espressif/iot_bridge (0.11.3) requires espressif/usb_host_cdc_acm (2.*) or
> >   espressif/esp_modem_usb_dte (1.1.0~1).
> > 
> >   And because no versions of espressif/iot_bridge match >=0.0.0,<0.11.3 ||
> >   >0.11.3
> > 
> >    and espressif/esp_modem_usb_dte (1.1.0~1) depends on espressif/usb_host_cdc_acm (2.*), espressif/iot_bridge (>=0.0.0) requires espressif/usb_host_cdc_acm (>=2.0.0,<3.0.0).
> > 
> >   So, because no versions of espressif/usb_host_cdc_acm match 2.*
> > 
> >    and project depends on espressif/iot_bridge (*), version solving failed.
> > 
> > Call Stack (most recent call first):
> >   /opt/esp/idf/tools/cmake/project.cmake:476 (idf_build_process)
> >   CMakeLists.txt:6 (project)
> > 
> > 
> > -- Configuring incomplete, errors occurred!
> > See also "/projects/esp-iot-bridge/examples/4g_hotspot/build/CMakeFiles/CMakeOutput.log".
> > cmake failed with exit code 1
> > ```
> 
> You can roll back idf-component-manager to version 1.4.1 and give it a try. `pip install idf-component-manager==1.4.1`

This worked!! and i was able to compile and flash the 4g_hotspot example. thank you

benjamin-sl, 2024-05-03:
> Are you communicating with the 4G module using UART? UART typically has low data rates. Could the low data rate be causing issues with OTA updates, or do you need to confirm whether the data can actually access the external network?

I am communicating via the UART. the low data rate would be the issue. 
THe other ESP32 is able to communicate with the blynk server and publish values online at a decent rate required for an IoT device, but it is unable to perform an OTA.

BenjaminDeHeisenberg, 2024-05-03:
> You can roll back idf-component-manager to version 1.4.1 and give it a try. `pip install idf-component-manager==1.4.1`

I was able to compile and flash the code with your suggestion. 
thank you

BenjaminDeHeisenberg, 2024-05-03:
> Are you communicating with the 4G module using UART? UART typically has low data rates. Could the low data rate be causing issues with OTA updates, or do you need to confirm whether the data can actually access the external network?

Yes, The 4G module is connected using the UART. 
The low data rate would be the issue for the OTA not to happen as the esp32 can communicate with the external network at a low data rate to publish essential values when needed.