espressif/esp-iot-bridge #74: 使用wireless_nic遇到2个问题

linhuikui, 2024-05-10:
1、通过网页配置连接AP时，如果输入的AP密码错误，就不能再扫描AP了，点击扫描时，提示错误；log显示一直在连接AP，能不能设置重试多少次之后暂停？
2、打开了usb和ap共享网络模式，两个接口获取到的网络段都不一样，和无线路由器那边的IP段也不一样。我使用到的场景是固定IP通信的，设备端的IP和无线路由器不在同一网段，没办法通信。请问要如何解决？

tswen, 2024-05-10:
1. 可以将 BRIDGE_STATION_CANCEL_AUTO_CONNECT_WHEN_DISCONNECTED 配置项设置为 false，或者也可以更改 bridge_wifi.c 代码来设置重试多少之后暂停重连
2. 对于 bridge 方案来说，每个 dataforwarding netif（softap 和 usb）的网段都是不一样的，其可以直接向路由器下面的设备建立连接以及发起通信。

linhuikui, 2024-05-11:
> 2\. 对于 bridge 方案来说，每个 dataforwarding netif（softap 和 usb）的网段都是不一样的，其可以直接向路由器下面的设备建立连接以及发起通信。

对于第2点，想把esp32s2当成usb无线网卡使用，bridge方案，感觉无线路由器的数据没有直接传到设备端，而是在esp32中转换了；这就导致设备端的IP和无线路由器的IP不在同一网段，不是同一局域网。我的场景是局域网内直接固定IP通信，中间没有服务器的，也不需要连外网。实测设备端不能向上级设备发数据。

tswen, 2024-05-13:
因为 bridge 不是透传方案，所以 usb 或者 softap 连接的设备端都是拿的是 esp32 分配的 “内网” IP，不过其是可以直接 ping 通路由器以及路由器局域网下的其他设备，也可以正常建立连接，只不过路由器以及其下别的设备是无法直接向连接 usb 或者 softap 的设备端直接发起通信。

linhuikui, 2024-05-14:
那目前有透传的方案吗？

tswen, 2024-05-27:
没有