espressif/esp-iot-bridge #76: esp-iot-bridge里面工程编译一直报错

yangjinyiyiyi, 2024-06-12:
idf安装的最新版本5.1.4和5.2.2，esp-iot-bridge也clone最新代码，都尝试编译始终无法成功，没有一个工程能编译成功
[image]
git浏览器访问也正常，下载东西也正常。麻烦看看是什么问题

mofa1, 2024-06-15:
1.进入乐鑫组件库的网址https://components.espressif.com/
2.搜索tinyusb
3.选择第一个进入4.将组件添加到工程中
5.重新编译,可以解决你的问题,但是我编译到最后还会出现其他的问题。

yangjinyiyiyi, 2024-06-17:
[image]
你们编译的时候是不是开了代理，我这边没有这个东西，感觉是编译的时候要从git拉东西，拉不下来。

mofa1, 2024-06-17:
我也拉取不下来，我把自带的yaml文件里面的内容删除掉，然后去乐鑫组件库里面搜索同样与当前依赖名称相同的进行添加。乐鑫组件库的用法可以在网上搜索一下，挺详细的，我表达能力有限说不清楚。

yangjinyiyiyi, 2024-06-17:
[image]
[image]
我在The ESP Component Registry里只能找到iot_bridge，而需要的另外三个wifi_prov_mgr，web_server，usb_device无法找到添加进去.

yangjinyiyiyi, 2024-06-17:
[image]
把main文件夹下idf_component.yml改成这样，同时删除掉components/wifi_prov_mgr和components/usb/usb_device里的*.yml文件终于可以了。

mofa1, 2024-06-17:
👍

mofa1, 2024-06-17:
👍