espressif/esp-iot-bridge #78: ESP-IOT-Bridge编译失败

zyq20170820, 2024-06-24:
我用的是ESP32，使用了IDF版本5.2，5.1.4都编译失败了。使用的例程是wireless_nic。

tswen, 2024-06-24:
编译失败的输出是什么呢？

在终端中执行下 pip install idf-component-manager==1.4.1，将 idf-component-manager 固定到 1.4.1 版本再试试

zyq20170820, 2024-06-24:
已经固定了组件版本号为1.4.1，编译告警如下：
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 E:\test\esp-iot-bridge-master\examples\wireless_nic"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- git rev-parse returned 'fatal: detected dubious ownership in repository at 'C:/Espressif/frameworks/esp-idf-v5.2.1'
'C:/Espressif/frameworks/esp-idf-v5.2.1' is owned by:
        'S-1-5-32-544'
but the current user is:
        'S-1-5-21-1515137862-3536264409-990725638-1001'
To add an exception for this directory, call:

        git config --global --add safe.directory C:/Espressif/frameworks/esp-idf-v5.2.1'
fatal: detected dubious ownership in repository at 'C:/Espressif/frameworks/esp-idf-v5.2.1'
'C:/Espressif/frameworks/esp-idf-v5.2.1' is owned by:
        'S-1-5-32-544'
but the current user is:
        'S-1-5-21-1515137862-3536264409-990725638-1001'
To add an exception for this directory, call:

        git config --global --add safe.directory C:/Espressif/frameworks/esp-idf-v5.2.1
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at C:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/build.cmake:544 (message):
  ERROR: Component "usb_host_cdc_acm" does not support target esp32

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "E:/test/esp-iot-bridge-master/examples/wireless_nic/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

zyq20170820, 2024-06-24:
请问一下，esp32怎么去掉这个组件usb_host_cdc_acm。

tswen, 2024-06-27:
将 build 和 managed_components 和 dependencies.lock 都删了重新 set-target 试试。

https://github.com/espressif/esp-iot-bridge/blob/master/components/iot_bridge/idf_component.yml#L14-L17，usb_host_cdc_acm 是 esp_modem_usb_dte 依赖的组件，但是理论上 esp32 不会拉取 esp_modem_usb_dte 这个组件的。

zyq20170820, 2024-07-10:
esp-idf版本v5.2.1，设置目标为esp32s3时，配置报错。麻烦帮忙看看
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory E:\test\esp-iot-bridge-master\examples\wifi_router\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 E:\test\esp-iot-bridge-master\examples\wifi_router"...
-- Found Git: C:/Espressif/tools/idf-git/2.43.0/cmd/git.exe (found version "2.43.0.windows.1")
-- Component directory C:/Espressif/frameworks/esp-idf-v5.2.1/components/esp-thread-br does not contain a CMakeLists.txt file. No component will be added
-- git rev-parse returned 'fatal: detected dubious ownership in repository at 'C:/Espressif/frameworks/esp-idf-v5.2.1'
'C:/Espressif/frameworks/esp-idf-v5.2.1' is owned by:
        'S-1-5-32-544'
but the current user is:
        'S-1-5-21-1515137862-3536264409-990725638-1001'
To add an exception for this directory, call:

        git config --global --add safe.directory C:/Espressif/frameworks/esp-idf-v5.2.1'
fatal: detected dubious ownership in repository at 'C:/Espressif/frameworks/esp-idf-v5.2.1'
'C:/Espressif/frameworks/esp-idf-v5.2.1' is owned by:
        'S-1-5-32-544'
but the current user is:
        'S-1-5-21-1515137862-3536264409-990725638-1001'
To add an exception for this directory, call:

        git config --global --add safe.directory C:/Espressif/frameworks/esp-idf-v5.2.1
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Dependencies lock doesn't exist, solving dependencies.
NOTICE: Skipping optional dependency: usb_device
Using component placed at E:\test\esp-iot-bridge-master\components\iot_bridge for dependency espressif/iot_bridge(*), specified in E:/test/esp-iot-bridge-master/examples/wifi_router/main\idf_component.yml
.......................Using component placed at E:\test\esp-iot-bridge-master\components\iot_bridge for dependency espressif/iot_bridge(*), specified in E:/test/esp-iot-bridge-master/examples/wifi_router/main\idf_component.yml
CMake Error at C:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/build.cmake:544 (message):
  ERROR: Because no versions of espressif/iot_bridge match >=0.0.0,<0.11.8 ||
  >0.11.8

   and espressif/iot_bridge (0.11.8) depends on espressif/esp_modem_usb_dte (1.*), espressif/iot_bridge (>=0.0.0) requires espressif/esp_modem_usb_dte (1.*).

  So, because no versions of espressif/esp_modem_usb_dte match 1.*

   and project depends on espressif/iot_bridge (*), version solving failed.

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:6 (project)

.......................-- Configuring incomplete, errors occurred!
See also "E:/test/esp-iot-bridge-master/examples/wifi_router/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the E:\test\esp-iot-bridge-master\examples\wifi_router\build\log\idf_py_stderr_output_4940 and E:\test\esp-iot-bridge-master\examples\wifi_router\build\log\idf_py_stdout_output_4940
PS E:\test\esp-iot-bridge-master\examples\wifi_router>

zyq20170820, 2024-07-10:
我之前下载的代码，然后cd进目录编译有问题。用工具下载idf.py create-project-from-example "espressif/iot_bridge=*:wifi_router"编译是正常的，ipv6是不是不能用？我测试好像只有ipv4。连接方式：STA接路由器，网口接电脑。

tswen, 2024-07-12:
bridge 方案不支持 ipv6

zyq20170820, 2024-07-22:
请问一下如果需要使用ipv6，网络连接方式同wifi_router一样：STA接广域网路由器，网口接电脑/嵌入式设备。有其它例程推荐吗？

tswen, 2024-07-24:
https://github.com/espressif/esp-idf/tree/master/examples/network/eth2ap
可以仿照这个透传方案改一下，改成 sta2ap。

zyq20170820, 2024-07-25:
请教一下：https://github.com/espressif/esp-idf/tree/master/examples/network/eth2ap与https://github.com/espressif/esp-idf/tree/master/examples/network/sta2eth 这两个例程的区别，两个例程ipv4都是正常的，ipv6只有eth2ap是正常的。我看代码sta2eth在MAC地址上做了处理，只处理ipv4的MAC地址，如果ipv6也要考虑，请问一下sta2eth可以怎么修改呢？谢谢

tswen, 2024-07-25:
https://github.com/espressif/esp-idf/blob/master/examples/network/sta2eth/main/ethernet_iface.c#L153

那应该只支持 ipv4。