espressif/esp-iot-bridge #79: undefined reference to `ip_napt_table_clear'

taxidriver2022, 2024-07-06:
[image]
我在尝试编译时发现managed_components文件夹下esp-iot-bridge中新增了一个意义不明的函数，没有函数定义，导致我的项目无法编译通过，请尽快解决

taxidriver2022, 2024-07-10:
5.1.4

tswen, 2024-07-10:
建议将 build 删除之后重新编译一下。

因为编译时会自动给 idf 的 components/lwip/lwip 子模块打上一个 patch，你也可以去 idf 下确认下 https://github.com/espressif/esp-iot-bridge/blob/master/components/iot_bridge/patch/ip4_napt.patch

taxidriver2022, 2024-07-10:
[image]
自动编译生成了你说的这个文件，但是idf中的components/lwip/lwip 子模块好像没有打上patch，并且还是编译不过

taxidriver2022, 2024-07-12:
@tswen 

> 建议将 build 删除之后重新编译一下。
> 
> 因为编译时会自动给 idf 的 components/lwip/lwip 子模块打上一个 patch，你也可以去 idf 下确认下 https://github.com/espressif/esp-iot-bridge/blob/master/components/iot_bridge/patch/ip4_napt.patch

请问你说的打上patch是什么情况，这边编译时没有发现idf的改动

tswen, 2024-07-12:
可以根据 [ip4_napt.patch](https://github.com/espressif/esp-iot-bridge/blob/master/components/iot_bridge/patch/ip4_napt.patch)
修改内容，手动改一下 idf。

closed 2024-07-25 by taxidriver2022

lijiahua310, 2024-07-31:
请问楼主解决了吗？我也出现了这个问题