espressif/esp-iot-bridge #80: Problem with the wifi_router example

Gery485, 2024-07-08:
# Wifi_router

## Problem description
Aftern I tried to configure the example in different ways (different esp-idf verions: 5.2.2, 5.1.4, 5.3.0; changing RMII clock mode, etc.). I was always able to connect the device with the router, using BLE provisioning app of the SSID and PWD. But it came nothing else. I wanted to use the Ethernet-Interface mainly, but I didn´t get any access from the "bridge". Not even the SoftAP provisioning worked, that I wanted to use to check, if the provisioning interface between the end-device and ESP-modul comes.

## General description

Modul - WT32-ETH01
Target - ESP32 - WROVER KIT 3.3V

### PHY
SMI MDC GPIO - (23)
SMI MDIO GPIO - (18)
PHY Reset GPIO - (-1)
PHY Address - (1)

### Changes
- outcommenting the override_path in idf_component.yml
- outcommenting the button releated code-parts in app_main.c, making it working without an extra button

## Use-Case
I want to build a bridge that can the following:
router --WiFi--> ESP-modul --Ethernet--> End-device
It´s because I have a device that only has ethernet, but want to use it without extra ethernet cable.

tswen, 2024-07-09:
Hello, you need to enable the BRIDGE_DATA_FORWARDING_NETIF_ETHERNET configuration item in menuconfig to enable Ethernet netif.

Gery485, 2024-07-09:
Hi @tswen, I did it, but it doesn´t work yet, is there anything else (special) I should do?

tswen, 2024-07-10:
Can you provide some logs?

Gery485, 2024-07-10:
Yes, that´s everything I get after monitoring the built/flashed project:
```
load:0x40080404,len:3876
entry 0x4008064c
I (33) boot: ESP-IDF v5.1.4-435-ga06c6877fa-dirty 2nd stage bootloader
I (33) boot: compile time Jul 10 2024 15:10:42
I (35) boot: Multicore bootloader
I (39) boot: chip revision: v1.1
I (43) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (80) boot:  1 phy_init         RF data          01 01 00015000 0000b000
I (87) boot:  2 ota_0            OTA app          00 10 00020000 001d0000
I (95) boot:  3 ota_1            OTA app          00 11 001f0000 001d0000
I (102) boot: End of partition table
I (107) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=3cf8ch (249740) map
I (205) esp_image: segment 1: paddr=0005cfb4 vaddr=3ffbdb60 size=03064h ( 12388) load
I (211) esp_image: segment 2: paddr=00060020 vaddr=400d0020 size=d3e54h (867924) map
I (524) esp_image: segment 3: paddr=00133e7c vaddr=3ffc0bc4 size=02cb4h ( 11444) load
I (529) esp_image: segment 4: paddr=00136b38 vaddr=40080000 size=1dce4h (122084) load
I (596) boot: Loaded app from partition at offset 0x20000
I (596) boot: Disabling RNG early entropy source...
I (608) cpu_start: Multicore app
I (608) cpu_start: Pro cpu up.
I (608) cpu_start: Starting app cpu, entry point is 0x400814f0
0x400814f0: call_start_cpu1 at C:/esp32/5.1/esp-idf/components/esp_system/port/cpu_start.c:159

I (0) cpu_start: App cpu up.
I (626) cpu_start: Pro cpu start user code
I (626) cpu_start: cpu freq: 240000000 Hz
I (626) cpu_start: Application information:
I (631) cpu_start: Project name:     wifi_router
I (636) cpu_start: App version:      v1.0.0-318-g7314e05-dirty
I (643) cpu_start: Compile time:     Jul 10 2024 15:10:03
I (649) cpu_start: ELF file SHA256:  5e37564b496cecde...
I (655) cpu_start: ESP-IDF:          v5.1.4-435-ga06c6877fa-dirty
I (661) cpu_start: Min chip rev:     v0.0
I (666) cpu_start: Max chip rev:     v3.99 
I (671) cpu_start: Chip rev:         v1.1
I (676) heap_init: Initializing. RAM available for dynamic allocation:
I (683) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (689) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (695) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (701) heap_init: At 3FFCA3F0 len 00015C10 (87 KiB): DRAM
I (707) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (714) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (720) heap_init: At 4009DCE4 len 0000231C (8 KiB): IRAM
I (728) spi_flash: detected chip: generic
I (731) spi_flash: flash io: dio
I (735) esp_core_dump_flash: Init core dump to flash
E (741) esp_core_dump_flash: No core dump partition found!
E (747) esp_core_dump_flash: No core dump partition found!
I (753) coexist: coex firmware version: d96c1e51f
I (759) app_start: Starting scheduler on CPU0
  (763) app_start: Starting scheduler on CPU1
mI (763) main_task: Started on CPU0
I (772) main_task: Calling app_main()
I (808) bridge_common: esp-iot-bridge version: 0.11.8
I (810) wifi:wifi driver task: 3ffcfce8, prio:23, stack:6656, core=0
I (824) wifi:wifi firmware version: 0d8e81df4
I (824) wifi:wifi certification version: v7.0
I (824) wifi:config NVS flash: enabled
I (824) wifi:config nano formating: disabled
I (829) wifi:Init data frame dynamic rx buffer num: 32
I (834) wifi:Init static rx mgmt buffer num: 5
I (837) wifi:Init management short buffer num: 32
[... 185 lines cut ...]

    esp_bridge_create_all_netif();

#if defined(CONFIG_BRIDGE_DATA_FORWARDING_NETIF_SOFTAP)
    wifi_config_t wifi_cfg = {
        .ap = {
            .ssid = CONFIG_BRIDGE_SOFTAP_SSID,
            .password = CONFIG_BRIDGE_SOFTAP_PASSWORD,
        }
    };
    esp_bridge_wifi_set_config(WIFI_IF_AP, &wifi_cfg);
#endif
#if defined(CONFIG_BRIDGE_EXTERNAL_NETIF_STATION)
    esp_wifi_connect();
#endif

#if defined(CONFIG_APP_BRIDGE_USE_WEB_SERVER)
    StartWebServer();
#endif 
#if defined(CONFIG_APP_BRIDGE_USE_WIFI_PROVISIONING_OVER_BLE)
    esp_bridge_wifi_prov_mgr();
#endif
}
```
Should I also provide some special logs?

tswen, 2024-07-12:
Which ETH PHY are you using? Have you tested it with the ethernet/basic example below from idf?

Gery485, 2024-07-15:
Jape, I tried it and it seem to be working. I use the WT32-ETH01 board, so the ethernet modul is a build in one. The already mentioned PHY pinouts should be alright.

adityanijave, 2024-07-19:
hey, i am facing the same issue let me know if you get a solution.

Gery485, 2024-10-03:
Hi @tswen, is there any solution already?
Should I may use another microcontroller for that specific task?

tswen, 2024-10-09:
https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32/esp32-ethernet-kit/index.html
I can successfully assign IP regardless of whether I use ESP32+IP101 or SPI Ethernet modules such as W5500, DM9051, etc. At first, I suspected that it was a hardware problem, but you mentioned that you can run the Ethernet example under idf, so I am more confused. Do you have other ESP32 boards with Ethernet?