espressif/esp-iot-bridge #81: Problem trying to build esp-iot-bridge examples

imfatant, 2024-07-14:
I've been using VSCode under Windows with the ESP-IDF framework for a while, but I'm now trying out esp-iot-bridge. After using `git clone https://github.com/espressif/esp-iot-bridge` I cannot build any of the examples. When I try to select a target board, I get a whole bunch of errors. 

Where am I supposed to put the esp-iot-bridge directory in relation to the ESP-IDF? And are there any Windows-specific environment variables that have to be set?

Thanks :)

tswen, 2024-07-15:
Please execute `pip install idf-component-manager==1.4.1`

imfatant, 2024-07-17:
Hi,

I did that, and I'm afraid I'm still having the same problem. Here is the output of `pip install idf-component-manager==1.4.1`:
```
PS C:\Users\monster\esp\esp-iot-bridge\examples\wireless_nic> pip install idf-component-manager==1.4.1
Collecting idf-component-manager==1.4.1
  Downloading idf_component_manager-1.4.1-py2.py3-none-any.whl.metadata (11 kB)
Requirement already satisfied: packaging in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (24.1)
Requirement already satisfied: requests<3 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (2.32.3)
Requirement already satisfied: urllib3<2 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (1.26.19)
Requirement already satisfied: requests-file in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (1.5.1)
Requirement already satisfied: requests-toolbelt in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (1.0.0)
Requirement already satisfied: schema in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (0.7.5)
Requirement already satisfied: six in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (1.16.0)
Requirement already satisfied: tqdm<5 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (4.66.4)
Requirement already satisfied: pyparsing in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (3.1.2)
Requirement already satisfied: colorama in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (0.4.6)
Requirement already satisfied: pyyaml>5.2 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (6.0.1)
Requirement already satisfied: cachecontrol>0.12.6 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from cachecontrol[filecache]>0.12.6; python_version >= "3.6"->idf-component-manager==1.4.1) (0.14.0)
Requirement already satisfied: contextlib2>0.6.0 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (21.6.0)
Requirement already satisfied: click in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from idf-component-manager==1.4.1) (8.1.7)
Requirement already satisfied: msgpack<2.0.0,>=0.5.2 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from cachecontrol>0.12.6->cachecontrol[filecache]>0.12.6; python_version >= "3.6"->idf-component-manager==1.4.1) (1.0.8)
Requirement already satisfied: filelock>=3.8.0 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from cachecontrol[filecache]>0.12.6; python_version >= "3.6"->idf-component-manager==1.4.1) (3.15.3)
Requirement already satisfied: charset-normalizer<4,>=2 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from requests<3->idf-component-manager==1.4.1) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from requests<3->idf-component-manager==1.4.1) (3.7)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\monster\.espressif\python_env\idf5.2_py3.11_env\lib\site-packages (from requests<3->idf-component-manager==1.4.1) (2024.6.2)
Downloading idf_component_manager-1.4.1-py2.py3-none-any.whl (140 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 140.7/140.7 kB 4.2 MB/s eta 0:00:00
Installing collected packages: idf-component-manager
  Attempting uninstall: idf-component-manager
    Found existing installation: idf-component-manager 1.5.3
    Uninstalling idf-component-manager-1.5.3:
      Successfully uninstalled idf-component-manager-1.5.3
Successfully installed idf-component-manager-1.4.1
```

And here is the output of `idf.py menuconfig`:
```
PS C:\Users\monster\esp\esp-iot-bridge\examples\wireless_nic> idf.py menuconfig
Executing action: menuconfig
Running cmake in directory C:\Users\monster\esp\esp-iot-bridge\examples\wireless_nic\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Users\monster\.espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Users\monster\esp\esp-iot-bridge\examples\wireless_nic"...
-- Found Git: C:/Users/monster/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/monster/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/monster/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
[... cut ...]
  HINT: Please check manifest file of the following component(s): main

  ERROR: Because project depends on espressif/button (2.*) which doesn't
  match any versions, version solving failed.

Call Stack (most recent call first):
  C:/Users/monster/esp/v5.2.2/esp-idf/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/monster/esp/esp-iot-bridge/examples/wireless_nic/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

adityanijave, 2024-07-19:
have you got the solution of this issue?

imfatant, 2024-07-20:
No...

imfatant, 2024-08-04:
I a no closer to being able to compile any of these esp-iot-bridge examples. I have read all the guides. Could someone just help? PLEASE?

tswen, 2024-08-09:
Hi, If you've already ensured that the `idf-component-manager` version is 1.4.1 and it's still not compiling, it's likely a network environment issue or an IDF environment problem. Do you have a Linux system available to test it on your side? Alternatively, you could try uninstalling and reinstalling the ESP-IDF environment on your Windows system.