espressif/esp-iot-bridge #82: 请问这个项目要在idf哪个版本上运行？

yunfusheng33, 2024-07-18:
我试图参照wireless_nic里面的代码，但是里面许多函数找不到定义，如
esp_netif_up(netif);
esp_netif_get_handle_from_netif_impl(netif);

tswen, 2024-07-18:
https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/User_Guide_CN.md#%E7%AE%80%E4%BB%8B

yunfusheng33, 2024-07-19:
我应该如何构建工程？我现在是把主目录下example和testapp删除了，并把wireless_nic里的内容拷贝到主目录下做main运行，但这样似乎不对

tswen, 2024-07-19:
1. 直接基于原始的 esp-mesh-lite 工程里面的 examples/mesh_local_control 进行开发
2. 执行 `idf.py create-project-from-example "espressif/mesh_lite^1.0.0:mesh_local_control"` 命令创建一个 mesh_local_control 工程
3. 将 esp-mesh-lite 工程里面的 examples/mesh_local_control 示例复制一份到自己的某个目录，直接编译也可以

wuyuanyi135, 2024-07-20:
@tswen 文档说能支持5.2 5.3 但是实际上tinyusb现在用的内容只能用到5.2.0 再向上esp-idf的usb重构导致无法编译了。

tswen, 2024-07-22:
是可以正常编译运行的，但是目前只能使用 NCM。

tswen, 2024-07-23:
没有，5.2 以后 USB Net Class 除了暂时不支持 Windows，其他都挺好的

wuyuanyi135, 2024-07-23:
@tswen 5.2之后的tinyusb不是自带的是ncm吗，这个应该windows是原生支持的？

Gary-404notfound, 2024-09-29:
> 没有，5.2 以后 USB Net Class 除了暂时不支持 Windows，其他都挺好的

我就说为啥配置成TINYUSB_NET_MODE_ECM_RNDIS一直无法通过编译，是因为版本过高了啊