espressif/esp-iot-bridge #83: T_INTER_COM -

GeminiServer, 2024-07-23:
I Try to add the following device. But i'm stucking on the ETH. 
I did the menuconfig and tried to set every possible setting to get a working **"4G NIC example"**
After uploading i will get the following error and the device reboots in loop. Seems that the ETH is not coming up correctly. i Hope some can help me out here. Thanks...

```
....
E (624) esp.emac: emac_esp32_init(426): reset timeout
E (624) esp_eth: esp_eth_driver_install(228): init mac failed
E (625) esp_eth: esp_eth_increase_reference(524): ethernet driver handle can't be null
E (634) esp_eth: esp_eth_update_input_path(332): ethernet driver handle can't be null
E (642) esp_eth: esp_eth_ioctl(400): ethernet driver handle can't be null
I (650) esp_eth.netif.netif_glue: 70:de:0d:40:04:41
I (655) esp_eth.netif.netif_glue: ethernet attached to netif
E (661) esp_eth: esp_eth_start(282): ethernet driver handle can't be null
....
```

Here the sdkconfig: 

[sdkconfig.txt](https://github.com/user-attachments/files/16348870/sdkconfig.txt)

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (31) boot: ESP-IDF v5.4-dev-1523-gaf25eb447e-dirty 2nd stage bootloader
I (31) boot: compile time Jul 23 2024 14:32:52
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v3.1
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (79) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  3 factory          factory app      00 00 00010000 00100000
I (101) boot:  4 coredump         Unknown data     01 03 00110000 00010000
I (109) boot:  5 ota_0            OTA app          00 10 00120000 00100000
I (116) boot:  6 ota_1            OTA app          00 11 00220000 00100000
I (124) boot: End of partition table
I (128) boot: Defaulting to factory image
I (133) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=18adch (101084) map
I (176) esp_image: segment 1: paddr=00028b04 vaddr=3ffb0000 size=02a38h ( 10808) load
I (180) esp_image: segment 2: paddr=0002b544 vaddr=40080000 size=04ad4h ( 19156) load
I (189) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=488cch (297164) map
I (292) esp_image: segment 4: paddr=000788f4 vaddr=40084ad4 size=0ab94h ( 43924) load
I (317) boot: Loaded app from partition at offset 0x10000
I (317) boot: Disabling RNG early entropy source...
I (329) cpu_start: Multicore app
I (338) cpu_start: Pro cpu start user code
I (338) cpu_start: cpu freq: 240000000 Hz
I (338) app_init: Application information:
I (341) app_init: Project name:     4g_nic
I (345) app_init: App version:      v1.0.0-318-g7314e05
I (351) app_init: Compile time:     Jul 23 2024 14:32:51
I (357) app_init: ELF file SHA256:  96e56bcc0...
I (363) app_init: ESP-IDF:          v5.4-dev-1523-gaf25eb447e-dirty
I (370) efuse_init: Min chip rev:     v0.0
I (374) efuse_init: Max chip rev:     v3.99
I (379) efuse_init: Chip rev:         v3.1
I (384) heap_init: Initializing. RAM available for dynamic allocation:
I (392) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (397) heap_init: At 3FFB4098 len 0002BF68 (175 KiB): DRAM
I (404) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (410) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (416) heap_init: At 4008F668 len 00010998 (66 KiB): IRAM
I (424) spi_flash: detected chip: gd
I (427) spi_flash: flash io: dio
W (431) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (444) esp_core_dump_flash: Init core dump to flash
I (450) esp_core_dump_flash: Found partition 'coredump' @ 110000 65536 bytes
I (472) esp_core_dump_flash: Core dump data checksum is correct
I (472) esp_core_dump_flash: Found core dump 14372 bytes in flash @ 0x110000
I (475) main_task: Started on CPU0
I (478) main_task: Calling app_main()
I (506) bridge_common: esp-iot-bridge version: 0.11.8
I (507) bridge_common: IP Address:192.168.4.1
I (507) bridge_common: GW Address:192.168.4.1
I (511) bridge_common: NM Address:255.255.255.0
I (516) mac select: MAC 64:b7:08:2b:fa:f4
E (624) esp.emac: emac_esp32_init(426): reset timeout
E (624) esp_eth: esp_eth_driver_install(228): init mac failed
E (625) esp_eth: esp_eth_increase_reference(524): ethernet driver handle can't be null
E (634) esp_eth: esp_eth_update_input_path(332): ethernet driver handle can't be null
E (642) esp_eth: esp_eth_ioctl(400): ethernet driver handle can't be null
I (650) esp_eth.netif.netif_glue: 70:de:0d:40:04:41
I (655) esp_eth.netif.netif_glue: ethernet attached to netif
E (661) esp_eth: esp_eth_start(282): ethernet driver handle can't be null
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400dd46f  PS      : 0x00060c30  A0      : 0x80107d30  A1      : 0x3ffb9190
0x400dd46f: esp_eth_transmit at /Users/ecolak/esp/esp-idf/components/esp_eth/src/esp_eth.c:344

A2      : 0x00000000  A3      : 0x3ffc423e  A4      : 0x0000002a  A5      : 0x3ffbacac
A6      : 0x00000000  A7      : 0x0000003c  A8      : 0x800820c0  A9      : 0x3ffb9180
A10     : 0x3ffc422c  A11     : 0x3ffc423e  A12     : 0x0000002a  A13     : 0x3ffb9364
A14     : 0x3ffb6820  A15     : 0x0000003c  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
0x4000c2e0: memcpy in ROM
0x4000c2f6: memcpy in ROM

Backtrace: 0x400dd46c:0x3ffb9190 0x40107d2d:0x3ffb91c0 0x40108509:0x3ffb91f0 0x400ff41d:0x3ffb9210 0x400f9519:0x3ffb9230 0x400f9549:0x3ffb9260 0x400f96c8:0x3ffb9290 0x400f0962:0x3ffb92b0 0x400f105c:0x3ffb92d0 0x40107921:0x3ffb9300 0x40106c1a:0x3ffb9320 0x400f0231:0x3ffb9340 0x400f02d5:0x3ffb9360 0x40088255:0x3ffb9390
0x400dd46c: esp_eth_transmit at /Users/ecolak/esp/esp-idf/components/esp_eth/src/esp_eth.c:344
0x40107d2d: esp_netif_transmit at /Users/ecolak/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1314
0x40108509: ethernet_low_level_output at /Users/ecolak/esp/esp-idf/components/esp_netif/lwip/netif/ethernetif.c:84
0x400ff41d: ethernet_output at /Users/ecolak/esp/esp-idf/components/lwip/lwip/src/netif/ethernet.c:312
0x400f9519: etharp_raw at /Users/ecolak/esp/esp-idf/components/lwip/lwip/src/core/ipv4/etharp.c:1172
0x400f9549: etharp_request_dst at /Users/ecolak/esp/esp-idf/components/lwip/lwip/src/core/ipv4/etharp.c:1199
0x400f96c8: etharp_request at /Users/ecolak/esp/esp-idf/components/lwip/lwip/src/core/ipv4/etharp.c:1217
0x400f0962: netif_issue_reports at /Users/ecolak/esp/esp-idf/components/lwip/lwip/src/core/netif.c:902
0x400f105c: netif_set_up at /Users/ecolak/esp/esp-idf/components/lwip/lwip/src/core/netif.c:876
0x40107921: esp_netif_up_api at /Users/ecolak/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1750
0x40106c1a: esp_netif_api_cb at /Users/ecolak/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:200
0x400f0231: tcpip_thread_handle_msg at /Users/ecolak/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:162
0x400f02d5: tcpip_thread at /Users/ecolak/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:148
0x40088255: vPortTaskWrapper at /Users/ecolak/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 96e56bcc0

I (758) esp_core_dump_flash: Save core dump to flash...
I (764) esp_core_dump_flash: Erase flash 16384 bytes @ 0x110000
I (948) esp_core_dump_flash: Write end offset 0x3824, check sum length 4
I (948) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...

```

An this is the device:
[image]

The IOs are:
```
#elif defined(LILYGO_T_INTER_COM)
#define ETH_TYPE                        ETH_PHY_LAN8720
#define ETH_ADDR                        0
#define ETH_CLK_MODE                    ETH_CLOCK_GPIO0_OUT
#define ETH_RESET_PIN                   4
#define ETH_MDC_PIN                     23
#define ETH_MDIO_PIN                    18
#define SD_MISO_PIN                     2
#define SD_MOSI_PIN                     15
#define SD_SCLK_PIN                     14
#define SD_CS_PIN                       13
#define MODEM_RX_PIN                    35
#define MODEM_TX_PIN                    33
#define MODEM_PWRKEY_PIN                32
#define RGBLED_PIN                      12

```

GeminiServer, 2024-07-25:
Now i'm getting the following error: 

`0) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1 `
Looks like a memleak or similar.. any ideas?

Here the whole log:
```

entry 0x40080640
I (31) boot: ESP-IDF v5.4-dev-1523-gaf25eb447e-dirty 2nd stage bootloader
I (31) boot: compile time Jul 24 2024 16:15:13
I (33) boot: Multicore bootloader
I (38) boot: chip revision: v3.1
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 16MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (79) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  3 factory          factory app      00 00 00010000 00100000
I (101) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (109) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (116) boot: End of partition table
I (121) boot: Defaulting to factory image
I (125) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=17184h ( 94596) map
I (166) esp_image: segment 1: paddr=000271ac vaddr=3ffb0000 size=02560h (  9568) load
I (170) esp_image: segment 2: paddr=00029714 vaddr=40080000 size=06904h ( 26884) load
I (183) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=46a34h (289332) map
I (282) esp_image: segment 4: paddr=00076a5c vaddr=40086904 size=08c84h ( 35972) load
I (304) boot: Loaded app from partition at offset 0x10000
I (304) boot: Disabling RNG early entropy source...
I (316) cpu_start: Multicore app
I (325) cpu_start: Pro cpu start user code
I (325) cpu_start: cpu freq: 160000000 Hz
I (325) app_init: Application information:
I (328) app_init: Project name:     4g_hotspot
I (333) app_init: App version:      v1.0.0-318-g7314e05-dirty
I (339) app_init: Compile time:     Jul 24 2024 16:17:14
I (345) app_init: ELF file SHA256:  915d8d2c9...
I (350) app_init: ESP-IDF:          v5.4-dev-1523-gaf25eb447e-dirty
I (357) efuse_init: Min chip rev:     v0.0
I (362) efuse_init: Max chip rev:     v3.99
I (367) efuse_init: Chip rev:         v3.1
I (372) heap_init: Initializing. RAM available for dynamic allocation:
I (379) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (385) heap_init: At 3FFB3890 len 0002C770 (177 KiB): DRAM
I (391) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (398) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (404) heap_init: At 4008F588 len 00010A78 (66 KiB): IRAM
I (412) spi_flash: detected chip: gd
I (415) spi_flash: flash io: dio
I (419) main_task: Started on CPU0
I (422) main_task: Calling app_main()
I (426) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (435) gpio: GPIO[32]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2068) bridge_common: esp-iot-bridge version: 0.11.8
I (2069) bridge_common: IP Address:192.168.4.1
I (2070) bridge_common: GW Address:192.168.4.1
I (2073) bridge_common: NM Address:255.255.255.0
I (2079) mac select: MAC 64:b7:08:2b:fa:f4
I (2110) esp_eth.netif.netif_glue: 64:b7:08:2b:fa:f7
I (2111) esp_eth.netif.netif_glue: ethernet attached to netif
I (3711) bridge_eth: Ethernet Started
I (3712) bridge_eth: Ethernet Link Up
I (3712) bridge_eth: Ethernet HW Addr 64:b7:08:2b:fa:f7
I (3715) bridge_eth: [ETH_LAN     ]
Add netif eth with 7314e05(commit id)
I (3722) bridge_common: netif list add success
I (3727) bridge_eth: ETH IP Address:192.168.4.1
W (3734) bridge_modem: Force reset 4g board
I (3737) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (3872) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (9249) uart: queue free spaces: 40
I (9274) bridge_modem: Signal quality: rssi=22, ber=99
I (9351) bridge_modem: Waiting for IP address
I (9712) bridge_modem: Modem Connect to PPP Server
I (9713) bridge_modem: ~~~~~~~~~~~~~~
I (9713) bridge_modem: IP          : 10.24.67.149
I (9713) esp-netif_lwip-ppp: Connected
I (9717) bridge_modem: Netmask     : 255.255.255.255
I (9727) bridge_modem: Gateway     : 10.64.64.64
I (9732) bridge_modem: Name Server1: 10.74.210.210
I (9738) bridge_modem: Name Server2: 10.74.210.211
I (9743) bridge_modem: ~~~~~~~~~~~~~~
I (9748) bridge_common: [ETH_LAN     ]Name Server1: 10.74.210.210
I (9754) bridge_modem: GOT ip event!!!
I (9748) main_task: Returned from app_main()
I (9759) bridge_modem: PPP state changed event 0
E (3494571) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (3497656) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (3500720) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
```

tswen, 2024-08-09:
`uart_task` is the task created by your application layer? You just need to make the task stack bigger.

ccie7009, 2024-08-09:
I did not use the GPIO0 for the RMII clock but used GPIO17 and it worked. I have a T-INTERNET-POE so it may be that that board uses a different GPIO for the RMII clock.