espressif/esp-iot-bridge #85: Wireless NIC device

maaamcube, 2024-09-16:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Hello everyone,
Does the wireless NIC works with usb devices ? i have an usb device on my stm32.
Thanks.

### Use cases

wireless NIC for usb devices

tswen, 2024-09-20:
You can select CONFIG_BRIDGE_DATA_FORWARDING_NETIF_USB to enable the USB Device function, which is possible if your stm32's USB Host supports USB Net Class.

maaamcube, 2024-09-20:
@tswen my stm32 can't work as host usb

tswen, 2024-09-20:
Then it is impossible to connect to stm32, because the USB interface of esp32-S implements Net Class and can only be used as a USB Device.

maaamcube, 2024-09-20:
@tswen usb host should be added.

tswen, 2024-09-23:
What is the purpose of connecting stm32 via USB? Is it used for CDC communication or something else? We support USB host. For specific examples of using USB host, please refer to:

- https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host
- https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host

maaamcube, 2024-09-23:
@tswen Yes, i'm using USB Vendor Specific communication

closed 2024-09-26 by maaamcube