espressif/esp-iot-bridge #87: Add usb host for wireless NIC

maaamcube, 2024-09-20:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Hello everyone,
I want to use my esp32 as usb host for my Stm32 usb device to send it through wifi with wireless NIC.
I submit this request to ask for usb host for esp32.
Thanks.

### Use cases

Send usb devices through wifi with wireless NIC

tswen, 2024-09-23:
iot bridge does not support USB host. If you want to use esp32 as USB host, please refer to：
- https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host
- https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host

maaamcube, 2024-09-23:
@tswen Yes, i'm asking if it's possible to add this feature in the future.

tswen, 2024-09-23:
What is the purpose of connecting stm32 via USB? Is it used for CDC communication or something else? We support USB host. For specific examples of using USB host, please refer to:

- https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host
- https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host

maaamcube, 2024-09-23:
@tswen I'm using USB vendor specific communication.

tswen, 2024-09-26:
USB Host is an independent functional component and does not need to be integrated into iot-bridge. If you want to use USB Host, refer to the example in the following link.
- https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host
- https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host
- https://components.espressif.com/components/espressif/usb_host_cdc_acm/versions/2.0.4

In addition, https://github.com/espressif/esp-iot-bridge/issues/85 is consistent with this issue. please close https://github.com/espressif/esp-iot-bridge/issues/85.