espressif/esp-iot-bridge #90: iot-bridge 是否支持充当软交换机作用

ZhMaio, 2024-10-29:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

我正在使用wifi_router示例进行运行，使用STA连接到局域网AP路由器上，然后在EPS32S3外接一个SPI接口的网卡（W5500），一切都配置完成，启动后STA连接路由器，然后PC机通过网线连接W5500，这是PC被分到的IP是192.168.5.2，我的局域网路由器是192.168.4.x，通过PC能访问192.168.4.x的设备，但是连接局域网的设备无法找到ESP32S3扩展的设备。
我想了解一下iot-bridge项目支持是否支持充当软交换机的功能？而且ESP32本身还能运行业务访问外部服务器
[image]

tswen, 2024-11-01:
不支持的，因为对于 PC1 来说，PC2 属于一个局域网内的设备，他是无法直接访问的。
不过可以通过 端口映射 的方式来简单实现一些对 PC2 的 ping 以及连接 PC2 上的 tcp server。
可以参考：
[portmap_guide.pdf](https://github.com/user-attachments/files/17597626/portmap_guide.pdf)

ZhMaio, 2024-11-01:
使用端口映射的方式解决了我的问题，我看到宣传适配中有对于BLE边界路由器的支持。在哪里能看到发布计划？

tswen, 2024-11-05:
后面有计划支持，但是优先级不高。后面支持了之后会在 bridge 仓库发布。

ZhMaio, 2024-11-05:
好的，了解了

closed 2024-11-05 by ZhMaio