espressif/esp-iot-bridge #91: Unable to see wifi AP when using wifi to wifi interface

GuillaumeHupi, 2024-11-06:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I use the latest version of esp-iot-bridge library. 
I use esp-idf V5.3.0. 
I use ESP32-Ethernet-Kit_V1.2

I use the "WIFI_ROUTER" example.
So as described in the User Guide. It should be possible do use wifi interface for both connecting to the external network and providing network data forwarding. 

So I configured my project like this : 
[image]
[image]

Here is my complete [sdkconfig file](https://github.com/user-attachments/files/17644270/sdkconfig.txt).

I did not modify the main code, only the sdkconfig. 

So what I expect after flashing this to my esp32 is a network called ESP_Bridge to appear. And what I see in the logs indicates that this should be the case :
```
I (1011) wifi:Disabled PMF config for STA
I (1015) bridge_wifi: [esp_bridge_wifi_set_config] softap ssid: ESP_Bridge password: 12345678
I (1197) wifi:Total power save buffer number: 16
I (1198) bridge_wifi: SoftAP config changed, deauth all station
I (1199) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.5.1
I (1209) button: IoT Button Version: 3.4.0
I (1211) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
start web server
```

### Actual behavior (suspected bug)

But I can not see this network appear with any of my devices (macbook or android phone). Even if I try to connect by typing manually ssid and password, It tells me that the network does not exists. 

I used the BLE APP to provide the SSID and PASSWORD for the ESP to connect to our wifi router. The esp connected successfully (I can see it with a network analyser). But I still can't see the ESP_Bridge network.

Full logs are added further down in the message.

I should point out that I have tried several configurations, including using Ethernet as the interface used to connect to the external network and WiFi as the interface used to provide network data forwarding for other devices, and in this case, I am able to see the ESP_Bridge network and connect to it. And everything works as expected.

It is only when WiFi is used for both interfaces that I do not see the network appear.

### Error logs or terminal output

```shell
load:0x40080404,len:3904
entry 0x40080640
I (31) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (31) boot: compile time Nov  4 2024 16:32:47
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v3.1
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (77) boot:  1 phy_init         RF data          01 01 00015000 0000b000
I (84) boot:  2 ota_0            OTA app          00 10 00020000 001d0000
I (92) boot:  3 ota_1            OTA app          00 11 001f0000 001d0000
[... 147 lines cut ...]
3. Modify the sdkconfig file [like this](https://github.com/user-attachments/files/17644270/sdkconfig.txt)

4. Flash the code to ESP32-Ethernet-Kit_V1.2

5. Try to connect to the wifi AP created

### Project release version

v1.0.0

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

macOS Sonoma 14.0

### Shell

ZSH

tswen, 2024-11-11:
Hello, please check this [part](https://github.com/espressif/esp-idf/blob/release/v5.3/components/wifi_provisioning/src/manager.c#L1535) in IDF. Could you confirm if this line of code sets the Wi-Fi mode to station mode?

Normally, when compiling the Mesh Lite example, the `manager.c` file is replaced by another [manager.c file](https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/src/wifi_prov/manager.c) located under the Mesh Lite component. This replacement prevents some provisioning logic from conflicting with the core logic of Mesh Lite.

GuillaumeHupi, 2024-11-12:
Hello, 

The wifi station part of my manager.c file located in "esp/v5.3/esp-idf/components/wifi_provisioning/src/manager.c" seems to be the same as the on that you linked : 

```
    /* Start Wi-Fi in Station Mode.
     * This is necessary for scanning to work */
    ret = esp_wifi_set_mode(WIFI_MODE_STA);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to set Wi-Fi mode to STA");
        goto err;
    }
```

I'm using the WIFI_ROUTER example from esp-iot-bridge library, not Mesh Lite. I don't know if the behavior is supposed to be the same. 

I found a way to make the WIFI_ROUTER example work. 
Firstly i kept the "Wi-Fi Provisioning over Bluetooth Low Energy." option checked in menuconfig. I provisioned the wifi credentials. So the ESP was able to connect to my wifi router with the provided credentials. But I still couldn't see the ESP'S AP appear, so I couldn't connect to it. 

So then I unchecked the "Wi-Fi Provisioning over Bluetooth Low Energy." option. And reflashed my ESP again, and this time I can connect to the ESP AP and the ESP automatically connected to my wifi router with the credentials kept in memory. So using it like this, it woks as expected, but it requires to disable wifi provisionning with BLE. 

Do you have any idea on how to make the example work with BLE provisioning activated ?

tswen, 2024-11-14:
Because idf's manager.c will set the wifi mode to WIFI_MODE_STA during network configuration, the AP is invalid at this time. You can comment out the idf code or add some logs to confirm.

ioviedodev, 2025-01-13:
Same issue