espressif/esp-iot-bridge #93: usb_cdc_4g_module连续重启

heyangxinmei, 2024-12-12:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

no

### Actual behavior (suspected bug)

[image]

### Steps to reproduce the behavior

alaways

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

windows10

### Shell

ZSH

tswen, 2024-12-13:
麻烦请在 https://github.com/espressif/esp-iot-solution 下提关于 
usb_cdc_4g_module

Sid9017, 2025-03-01:
问题解决了么？