espressif/esp-iot-bridge #94: How to access client ports

aHcVolle, 2025-01-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

For the last days I was trying to hook up my linux based edge device into my wifi network.
I did successfully compile the wireless_nic example, did setup the network access via my computer and can use the wireless_nic device to access the network. What I really want to do is to access the webinterface of the device I connect the wireless_nic to via USB (running on a ESP32S2) which unfortunately does not work.

When I first tried to access my clients webinterface I was presented the wifi AP setup page of the client. I then disabled the webserver on the client and hardcoded the wifi credentials in the client which I was hoping might help to clean up the ussage of port 80 only to find out that this did not really help.
I then tried to access the SSHD port of the client which did not work either so I guess I misunderstood the capabilites of the device. Or maybe this has to do something with the config. When connecting the client to a linux computer I can see a IP address of 192.168.4.2 even if the AP is configured as e.g. 192.168.5.102 via DHCP.

Is there any option I am missing or is the setup I want to achieve out of scope?

Best regards
 Chris

tswen, 2025-01-20:
Can you provide a topology diagram with icons representing the ESP32, PC, and wirelessly connected devices, and indicate which device the Web Server is on and which other device wants to access the Web Server?

xcguang, 2025-01-27:
Maybe you can try ip_portmap_add api to map your device from notebook to esp32.

aHcVolle, 2025-01-27:
Sounds like a good plan but I have no idea how I could do this. I guess this is no menuconfig switch i can easily toggle?

xcguang, 2025-01-27:
It needs you call ip_portmap_add in your code, for example, the server port on notebook is 8080, IP is 192.158.4.2, you can try ip_portmap_add(IP_PROTO_TCP, ipaddr_addr("192.168.4.1"), 8080, ipaddr_addr("192.168.4.2"), 8080), but I didnot test.
I'm afraid our guys can not help you in time due to Chinese Spring Festival Holiday.

aHcVolle, 2025-01-29:
Ok, thanks for the guidance. I'll try this as soon as I'm able to flash my boards again. (currently they somehow do not want to enter BL mode and i cannot flash them anymore :( ). Happy CNY!